robotology/icub-firmware-models #5: Publishing and documenting the model and its generated fw for coupling/decoupling eCub wrist.

valegagge, 2022-11-02:
Following the [creation of the wrist controller](https://github.com/icub-tech-iit/dev/issues/355) and its [integration in the ems fw](https://github.com/icub-tech-iit/dev/issues/358), we want to publish and document the model and its generated fw for coupling/decoupling of eCub wrist. 

Please see also this [integration](https://github.com/icub-tech-iit/dev/issues/458) activity.

ale-git, 2023-03-16:
Here it is the PR:
https://github.com/robotology/icub-firmware-models/pull/22

The folder also contains a README.md about the mathematics of the spherical joint.

pattacini, 2023-03-16:
Thank you @ale-git!

I'll be reviewing the PR asap.

pattacini, 2023-03-26:
Done via:
- https://github.com/robotology/icub-firmware-models/pull/26

closed 2023-03-26 by pattacini