robotology/icub-firmware-models #8: Make the `Vcc` and `Vmax` configurable via CAN

pattacini, 2023-02-19:
As of now, `Vcc` and `Vmax` are statically defined in the model. This prevents us from adapting those parameters to the setups/robots w/o re-generating the code.

It'd be thus great to have the possibility to tune the voltage values on the fly via CAN.
We didn't do that yet as the current CAN protocol doesn't foresee it.

pattacini, 2023-02-22:
Actually, it turns out that `Vcc` can be read by the board using ADC. This would enable us to automatically adapt wrt. a constantly changing battery voltage. Opening a dedicated issue for this.

pattacini, 2023-02-22:
cc @marcoaccame @sgiraz @fgarini @valegagge

valegagge, 2023-04-19:
Hi guys,
I just want to add a note regarding this new feature.

In `AMC-BLDC`, @ale-git had already implemented two motor parameters:
 - `swapBC`: it is a boolean that lets the user simulate the motor cable swap. This is useful in case of wrong cabling.
 - `hall sensor offset`

It would be great if these parameters were included in the new CAN protocol. 
We are available to discuss together where to insert these parameters.

I suggest only designing the new protocol taking in mind these parameters also, but not making them work: we can do it in a different activity.

pattacini, 2023-04-19:
Agreed @valegagge 👍🏻 
Added them up to the [OP](https://github.com/robotology/icub-firmware-models/issues/8#issue-1590612036).

mfussi66, 2025-08-12:
It might be useful to exhume this issue and finally attack it, since with the advent of the AMCFOC and the request of additional control parameters (e.g. https://github.com/robotology/icub-firmware/pull/618) require( an increase in configurability. 

Since we are still working with the existing CAN protocol, I propose the following:

**Parameters**

- `Vmax`: maximum voltage (average within pwm period) that can be applied to a motor
- `Hall offset`: offset in degrees between hall sensors and phases
- `Kbemf`: back emf constant used to compensate the former in FOC control
- `Kff`: Feedforward contribution in velocity control

In the icub can protocol, the first byte is reserved to the message class, so it will be ignored. The following 7 bytes could be:

| byte | Var | Type |
|--------|--------|--------|
| 1 | `Vmax` | uint8 |
| 2-3 | `Kbemf` | int16 |
| 4-5 | `Kff` | int16 |
| 6-7 | `shift_factor` | int16 |

The shift factor can scale kbemf or kff as usual, just like the PIDs, so their representation can fit inside the int16 if they are floats.
$$
K_{ff, shifted} = K_{ff} \cdot 2^{shift} 
$$

E.g. : 

$$
  \begin{gather}
 int16(0.03 \cdot 2^{10}) = 30 \\
30 \cdot 2^{-10} = 0.0293
  \end{gather}
$$

A bit of precision loss.

To handle the hall offset and swap, maybe we might benefit from another CAN message.

pattacini, 2025-08-12:
>If we save 3 bit for Vmax and exploit the spare trailing byte of the proposed packet above, we should end up with 8+3=11 bit for the Hall offset, which amounts to 2048 values to cover 360 degrees, meaning a precision of 0.17 deg. If we confirm that the latter precision is enough for the purpose of the Hall offset, I would definitely go with this solution, instead of churning out yet another CAN packet.

In this case, we should arrange the packet differently:

| byte | Var | Type |
|--------|--------|--------|
| 0 | ⚠️ reserved  |/ |
| 1-2 | `Kbemf` | int16 |
| 3-4 | `Kff` | int16 |
| 5 | `shift_factor` | int8 |
| 6-7 | `Hall offset` | uint11 |
| 7 | `Vmax`  | uint5 |

ale-git, 2025-08-22:
> To handle the hall offset and swap, maybe we might benefit from another CAN message. I think a Hall offset of 120 deg corresponds to a phase swap ([@ale-git](https://github.com/ale-git) correct me if i'm wrong), so maybe we dont need an explicit bool for `swapbc`. Though that could make the protocol less transparent.

Hi @mfussi66 , a 120 deg offset is a cyclic permutation, not a swap of two phases. There is no way you can obtain a mirroring from a rotation, and vice versa.

pattacini, 2025-08-28:
Nice!

mfussi66, 2025-09-04:
PR here: #146