robotology/icub-firmware-models #14: Model the `i2t` overcurrent protection

pattacini, 2023-02-22:
We're required to implement in the architectural model the [`i2t`][i2t] protection.
We may use some new parameters as per #8.

Some further resources:
- https://www.mathworks.com/help/sps/ref/fuse.html;jsessionid=85e939b80269c1d09f68330ff0fb

[i2t]: http://www.technosoft.ro/KB/index.php?/article/AA-00280/0/Over-Current-versus-i2t-protection-GB-CM.html

pattacini, 2023-03-14:
We may start off with a simple implementation where we just stick to the nominal current, w/o having access to those parameters that will need to be transmitted over CAN.

@mfussi66 can deal with this task, while I can act as a tutor.

pattacini, 2023-03-14:
Eventually, with the i2t implemented, we will be able to increase the current thresholds (overload, peak).

cc @maggia80 @sgiraz @Nicogene

pattacini, 2023-03-22:
Speaking with @maggia80, it turned out that Faulhaber is likely able to provide us with the missing parameters.

pattacini, 2023-03-23:
>Regarding our motor, the [datasheet](https://github.com/robotology/icub-firmware-models/files/11038670/EN_2214_BXTR_DFF.pdf) doesn't seem to contain all the required parameters, neither for the standard i2t (e.g., the time window) nor for the thermal model.

Actually, the online datasheet is more complete and contains the required information.
See https://www.faulhaber.com/en/products/series/2214bxtr/

[image]

Therefore, I think we can start, @mfussi66.
We need to:
- Meet to organize the work
- We will test the thermal model on a setup where we can take some thermal measurements.

pattacini, 2023-03-24:
A very quick sanity check with the values above:

$$
\begin{cases}
\begin{matrix}
R_L = 30 \Omega \\
R_{t1} = 7 K/W \\
R_{t2} = 49 K/W
\end{matrix}
\end{cases}.
$$

If we think of starting from the temperature of the environment $T_0 = 25$ $^oC$, then to get $\Delta T = 60$ $^oC$ to reach $85$ $^oC$ (way below $T_{crit}$) we should deliver:

$$
i = \sqrt{\frac{60 \cdot \left( 7+49\right)}{30 \cdot 7}} = 4 \text{A}.
$$

Well, 4 Amps is a bit off the motor specs.
There must be something else.

pattacini, 2023-03-26:
Moved to the next sprint.

pattacini, 2023-03-27:
>There must be something else 😕

A couple of comments:
- There might be something missing in our model that we can inquire about with the help of Faulhaber.
- We can fit the model with thermal measurements.

fiorisi, 2023-03-28:
> A very quick sanity check with the values above:
> 
> {RL=30ΩRt1=7K/WRt2=49K/W.
> 
> If we think of starting from the temperature of the environment T0=25 oC, then to get ΔT=60 oC to reach 85 oC (way below Tcrit) we should deliver:
> 
> i=60⋅(7+49)30⋅7=4A.
> 
> Well, 4 Amps is a bit off the motor specs. There must be something else 😕

If I correctly remember we have to balance the Joule losses, the dissipated power and the temperature change in the motor windings. The balance should be the following:

$P_j - P_d = C {\partial T_w\over dt}$ 

Where:
- $P_j$ are joule losses
- $P_d$ is the dissipated power
- $C$ is the thermal capacity
- $T_w$ is the temperature of the motor

This should become

$i^2 R_t - {{T_w - T_a} \over {R_{th1} + R_{th2} }} = C {\partial T_w\over dt}$

Where:
- $R_t$ is the terminal resistance
- $T_a$ is the ambient air temperature
- $R_th1$ is the thermal resistnce between the winding and the housing
- $R_th2$ is the thermal resistnce between the housing and the ambient air

We don't have $C$, but should be possible to compute it from the thermal time constant $\tau_{w1}$:

$C = \tau_{w1} \over {R_{th1} + R_{th2} }$

By solving the differential equation and initializing the widings temperature we should be able to check the model.

pattacini, 2023-03-30:
I did the calculations wrong!
I was basically confusing the heat source with a voltage generator rather than a current generator, in the analogy.

Here's the correct form:

$$
i = \sqrt{\frac{60}{30 \cdot \left( 7+49 \right)}} = 188 mA.
$$

The final outcome makes really sense to me.

pattacini, 2023-03-30:
### Notes for the development
We will need to accommodate a generic thermal model for a variety of motors. Hence, we should come up with a single equivalent thermal resistance and a single equivalent thermal time constant.

pattacini, 2023-03-30:
Let me attempt a recap after the notes above.

### 🎯 Objective
We aim to implement in our FW architecture (Simulink) a thermal model of the motor so that we can provide a further level of protection.
We should be able to adapt the model to a wide variety of motor brands.

### ⚙️ Implementation
For simplicity, we can start with a first-order thermal model, where the input heat flow $\dot{Q}$ is determined by the electrical current $i$[^1] and the resulting electrical resistance of the circuit $R_L$:

[^1]: We can refer to the quadrature current here, as the direct current is regulated to `0` by the PID.

$$
\dot{Q} = R_L \cdot i^2.
$$

The steady-state temperature variation wrt the environmental temperature $T_e$ is:

$$
\Delta T = \dot{Q} \cdot R_t,
$$

where $R_t$ is the thermal resistance, whose unit is K/W.

The temporal evolution of the temperature $T$ while evolving to $T_e+\Delta T$ is specified by the time constant $\tau$.

Therefore, it suffices to implement a digital filter $F$ characterized by the same constant $\tau$ and process the output $\hat{T}\left(t\right)=T_e+F\left(\Delta T\right)=T_e+F\left(R_L \cdot i^2 \cdot R_t\right)$.

Whenever $\hat{T}\left(t\right) \geq T_{crit}$, we will limit the current $i$ to the nominal value $I_N$.

For any $\dot{Q} \le \dot{Q}_N=R_L \cdot I^2_N$, the temperature will start decreasing.

Thereby, we may set a convenient threshold $T_L \le T_{crit}$ so that, as soon as $T \le T_L$, we disengage the thermal protection.

### 📝 Involved quantities
#### Datasheets
It seems that we have access to the quantities at stake from the [Faulhaber datasheet](https://www.faulhaber.com/en/products/series/2214bxth).

#### Parameters
We would need to define these parameters: $R_L$, $R_t$, $\tau$, $T_{crit}$, $T_L$.
By contrast, the nominal current $I_N$ is already accounted for in our architecture. 

#### Environmental temperature
We would need an estimate of $T_e$, unless we decide that we could assume $T_e = 25$ $^oC$ to start with.

pattacini, 2023-04-02:
Development started in https://github.com/robotology/icub-firmware-models/tree/14-amcbldc--model-the-i2t-overcurrent-pr.

mfussi66, 2023-04-03:
In continuous time it should result in the following model:

[image]

With 25.9 Ohms of terminal resistance the motor gets toasted at about 220mA.

pattacini, 2023-04-03:
We use those values to be able to perform measurements with the thermocamera. However, the motor won't get toasted at that temperature. 

For checking the motor's limits, we should switch to the other set of params corresponding to the windings.

At any rate, we can temporarily overcome the current limits as the temperature increases slowly compared to other dynamics at stake.

mfussi66, 2023-04-04:
In the digital domain with Tustin discretization and externally-defined coefficients, the filter could look like this:

$$
 \frac{T_s R_L R_T (z + 1) } {(T_s + 2 \tau) z + T_s - 2 \tau}
$$

Translated into configurable blocks: 

[image]

Comparing the continuous model with the discretized one, there is a small quantization error that stabilizes over a long time.

Here is the step response with 250mA. 

[image]

mfussi66, 2023-04-04:
I transferred the filter in the architectural model and generated the code, which seems very lean:

[image]

Additionally, the Keil project was updated and compiled, and is now ready to be tested.

pattacini, 2023-04-04:
Nice!

Nitpicking: in the comment we can see $R_{t1}$ and $R_{t2}$. Could you update it?

pattacini, 2023-04-04:
Also, I would make clearer that the coefficients depend on $T_s$.
So, $T_s$ may be a further input to the block `Compute coefficients`.

mfussi66, 2023-04-04:
> Also, I would make it clearer that the coefficients depend on Ts. So, Ts may be a further input to the block `Compute coefficients`.

> Nitpicking: in the comment we can see Rt1 and Rt2. Could you update it? We've got only Rt.

> ⚠️ The square block for the current might be missing.

Woops, silly oversight.  Will fix locally, and then put it in a fork after @pattacini checks why the thermal model code is embedded in a function that requires explicit call in the `mbdAgent`. It is undesired, since it would require a dedicated thread at 10ms.

pattacini, 2023-04-04:
> after @pattacini checks why the thermal model code is embedded in a function that requires explicit call in the `mbdAgent`. It is undesired, since it would require a dedicated thread at 10ms.

We will now generate two step functions:
- `step_Time_1ms`
- `step_Time_10ms`

that we need to call accordingly in the glue code.
No new thread at 10ms then, just a quick scheduler in the 1ms.

mfussi66, 2023-04-05:
Yesterday afternoon implemented the counter inside the `step_time_1ms()`, that on the 10th tick calls the `step_time_10ms()`. The thermal model runs, seemingly correctly, now it just needs proper testing.

mfussi66, 2023-04-06:
To work around the 4096 lines limitation of the debugger printf, I'll publish the temperature on the STATUS CAN packet temporarily.

pattacini, 2023-04-06:
As soon as we publish some nice graphs validating the working principle of the model, we can merge upstream against the feat branch.

I'd expect some fiddling to fit the model with the observations. However, this is a different point.

mfussi66, 2023-04-06:
Nonetheless, this is a first acquisition. There is an initial offset between the two measurements that is caused by a previous run of the wrist setup, I did not have time to let it cool down due to timing constraints, but we can clearly see that the difference between the two over time increases. 

[image]

here is the data: [test_i2t.zip](https://github.com/robotology/icub-firmware-models/files/11171123/test_i2t.zip)

pattacini, 2023-04-06:
At the moment, we're considering the quadrature current squared for simplicity (the direct current is `0`), but the assumption may be wrong, actually.

We can switch to the 3-phase currents but we need to agree on what kind of operation we should implement to replace the $i^2$ component.

mfussi66, 2023-04-07:
While reflecting on the usage of the 3-phase currents I checked again the discretized model and I noticed I made a mistake during some premature optimization, which caused the steady-state value to be significantly higher.

I took the measurements again with the fixed firmware, but sadly I did not have the tripod, so they are a little shaky. I filtered them with a median filter with a window of 3 samples, and the results are much much better:
[image]

📦: [i2t_test_2023_04_07.zip](https://github.com/robotology/icub-firmware-models/files/11178292/i2t_test_2023_04_07.zip)

Given this result, I would try to run more tests while using the $$I_q$$, and switch to the 3-phase currents as needed.

pattacini, 2023-04-07:
Came across a useful [reference][1]:

At page 14, one reads:

[image]

[1]: https://github.com/robotology/icub-firmware-models/files/11179365/RAJPUT-THESIS-2016.pdf

pattacini, 2023-04-07:
I reckon we can close this task.
In the upcoming sprints, we will address two more points:
- make observations in a wider range of temperature
- close the loop with the control logic

closed 2023-04-07 by pattacini