robotology/icub-firmware-models #18: AMCBLDC – Refactor the IRQ handler responsible for calling FOC

pattacini, 2023-02-22:
From @marcoaccame 

---

Another thought:

I am a little nervous to have an important high priority and high frequency IRQ Handler that executes for long time. The rule of thumb is: the more important the IRQ handler, the less code it must run. Now: it is executed every 37 us, lasts about 15 us. Also it runs code which requires mutual exclusion. A little too much in my opinion.

I would clean up and move out of the IRQ Handler all the MBD code by doing the following.

- keep inside the IRQ Handler only a quick copy from DMA area of the three currents (maybe into a circular FIFO) and use an RTOS event to wake up a very high priority thread called threadFOC.
- the high priority thread threadFOC is waken up by an event from the IRQ handler which: copies just the three currents in a thread safe and very quick mode (I may do it perhaps also w/out disabling the IRQ...) and then use the three currents to run the MBD FOC algorithm. 
- redefine the mutex used by the MBD code to be an RTOS mutex because it must protect data between two RTOS threads. The advantage is that the RTOS mutex has priority inversion.
- re-evaluate the duration and frequency of the IRQ Handler and of the two threads FOC and TICK.  
- if the IRQ Handler is still not regular, I would check the priorities of the IRQ handlers and of the DMA to make sure no other has higher priority.

---

The framework of the application is a little messy. It was developed long ago and it needs maybe some tidy up.

For now I document in here what we need to have.

[image]
**Figure**. Scheme of the required SW entities.

We need a new thread `tFOC` alongside to the existing threads `tCAN` and `tTICK`. Also we need a different mutual exclusion mechanism between actions performed inside `tTICK` and `tFOC`: we can use a `embot::os::rtos::mutex_t`. 

The interactions amongst entities are the following:
- `tCAN` -> `object`: when we have a new can frame from the `embot::hw::can` driver then thread `tCAN`, the producer, adds it to the input FIFO which is protected w/ a mutex vs concurrent access from `tTICK`, the consumer.
-  `tTICK` -> `tCAN`: if there is a can frame to send back as a reply or a periodic streaming the thread  `tTICK` sends an event to thread  `tCAN` which retrieves a vector of replies from the output FIFO (mutex protected).
-  `tTICK` <->  `tFOC`: access to shared data is done by means of a `rtos::mutex_t`. 
- `IRQ handler` ->  `tFOC`: the handler every 37 us has three new current values, it copies them into a volatile variable and then sends a wake up rtos event to thread  `tFOC`. The thread wakes up, disable IRQ for the handler, copies the volatile variable into local memory and re-enables IRQ. At this time it runs safely the FOC algorithm.

The  `tTICK` thread is periodic @ 1ms, the  `tFOC` is activated by the periodic IRQ handler @ 37us (so the thread is de-facto periodic as well).

Almost all objects are available. I just need to refactor and test.

marcoaccame, 2023-03-22:
I will wait the conclusion of tests on https://github.com/icub-tech-iit/study-mbd-amc-bldc/issues/258 before I proceed.

marcoaccame, 2023-03-31:
Some very quick pictures.
[image]
**Figure**. The `thrFOC` is called every about 37 us and it lasts about 15 us. We can see that it is triggered by the IRQ handler of the DMA1_CH2. The `thrFOC` is preempted only by the TIMER IRQ handler.

[image]
**Figure**. The `thrCTRL` is called every 1000 us. It lasts about 590 us despite being preempted by the `thrFOC` and by all the IRQ Handlers.

pattacini, 2023-04-03:
Agreed @marcoaccame 👍🏻 
Thanks for the final report!
Let's put this in Review.

closed 2023-04-04 by Nicogene