robotology/icub-firmware-models #34: AMCBLDC – Validate the thremal model further and link it to the control logic

pattacini, 2023-04-07:
Following up on #14, we're required to:
- Make more observations in a wider range of temperatures for validation purposes.
- Close the loop between the output of the thermal model and the control logic.

pattacini, 2023-04-10:
Regarding the control logic, we only need to fiddle with the supervisor_rx, as the newly created flag `enable_thermal_protection` is already handled by the outer controller.

mfussi66, 2023-04-18:
If we want to observe a wider spectrum of temperatures, I think we need to apply some kind of load to the wrist, or the motor on the Lego setup, otherwise we might never reach more than 35C, like in the previous test:
[image]

Also, we might benefit for some cooling tool like a fan, so that we can reach a zero temperature delta more quickly.

mfussi66, 2023-04-19:
@lawproto Helped me mount on the wrist the shaft that Dev used in the past, upon suggestion of @fiorisi :https://github.com/icub-tech-iit/ergocub-design-lowerarm/issues/106 

So now we can properly load it.

mfussi66, 2023-04-20:
At the moment the wrist setup is exhibiting a strange behaviour after mounting the weights' shaft.

There is a constant current $Iq = 45mA$ on the AMCBLDC 2 (out of 3) when the wrist is idle:

[image]

Even though the shaft weighs more than the 4-fingered hand, the joints do not tilt down because of gravity, so I expect the current to be 0 mA.

pattacini, 2023-04-20:
A bit puzzling.
We should understand the source of this readout in IDLE.

At any rate, in the meanwhile, we can start the tests disregarding this phenomenon.

mfussi66, 2023-04-20:
It seems like the problem is on the AMCBLDC 2, because the issue disappears if I switch to another board. Additionally, a noticeable ticking noise can be heard when the motor spins.

Nonetheless I agreed with @pattacini F2F that we can go on by monitoring another motor, eg the third one.

mfussi66, 2023-04-20:
I managed to start one decent trial, but the performance seems very poor on the model side. At 300 sec I added 500 gr weight, and even though the motor temperature increases noticeably, the model estimation shoots up.

A comment on the data:
- It needs temporal alignment, the dip in temperature estimation is caused by putting it in idle to load the weight
- The initial temperature that I removed as offset was 26C for the model, and 30C for the camera.

[image]

The only thing added was the $I_d^2$, correct? I need to think about why It could influence the estimation so much.

mfussi66, 2023-04-20:
That's what I thought too, we could log I'd through CAN as well and see how it behaves.

mfussi66, 2023-04-21:
The direct current $I_d$ is very noisy and it could have an effect on the overall evaluation of the temperature. It can reach high peaks of hundreds of milliamps, but it is quickly controlled to zero.

[image]

Here is a detail of the Id when the motor is moving. There are significant peaks, but they are not long lasting.

[image]

pattacini, 2023-04-21:
The thermal model is implemented through a LP filter with a very low pass-band; hence it should take the DC value (~ 0 A) and filters out all the high-frequency ripples.

At any rate, we may consider removing $I_d$ (temporarily[^1]) from the model to check the outcome.

[^1]: The direct current provides a contribution we cannot ignore.

mfussi66, 2023-04-21:
Reflecting on the parameters, the thermal resistance $R_{th2}=49 K/W$ seems worryingly high.

In steady state, applying constant $I_{q} = 200mA$, we get $\Delta T = 0.2^2 \cdot 25.9 \cdot (7 + 49) = 58C$, which is a temperature differential that was not observed so far, even with 25C ambient temperature.

It would be useful to know where 49 K/W comes from, because if we take 
- the rated current $I_N = 0.36 A$ at 22C ambient temperature
- the phase-phase resistance $R=25.9 \Omega$
- the maximum temperature allowed for the windings $T_{max}=125C$

We can find a total thermal resistance 

$$
\frac{\Delta T_{max}}{P_{max}} = \frac{125 - 22}{0.36^2 \cdot 25.9} = 30.7 K/W
$$

which is significantly lower than (7 + 49) K/W

If possible, I'd keep in touch with Faulhaber to gather more info, while we could fit the thermal model from the camera observations.

mfussi66, 2023-04-21:
Just for sake of completeness, here is the snippet that I use to call the model:
```cpp
    static uint8_t counter_Ts_10ms = 1;
    
    if(counter_Ts_10ms == 10)
    {
        AMC_BLDC_step_Time_10ms();
        counter_Ts_10ms = 1;
    }
    counter_Ts_10ms++;
```
Which at the moment seems ok to me and does not result in too many calls that would ruin the estimation.

pattacini, 2023-04-21:
```cpp
else counter_Ts_10ms++;
```
Otherwise the sample time will be 9 ms.

pattacini, 2023-04-22:
>Reflecting on the parameters, the thermal resistance $R_{th2}=49 K/W$ seems worryingly high. It could mean that a power draw of 2W results in 198C.

$2$ $W$ yields 98 $^oC$ over $R_{th2}=49 K/W$, not 198 $^oC$; was it a typo?
Our motor is of $6$ $W$ power and to reach $2$ $W$, we would thus need $i=\sqrt{2/25.9}=277$ $mA$, which is not implausible, after all.

Why are you relating $T_{max}$ to $I_N$?
$I_N$ doesn't cause the motor to reach $T_{max}$, in my understanding.

pattacini, 2023-04-24:
Something we kind of neglected so far:
Although $E\[I_d\]=0$, it comes out that $E\[I^2_d\]>0$ and thus it impacts the temperature estimation, as it should actually be.

mfussi66, 2023-04-27:
I made a couple of other tests during the past few days, both without and with added weights to the wrist. 

## No weight
This is the resulting plot without weight. Sadly, in the first case the thermal camera software crashed and the logging was lost, but I can guarantee that the measured temperature was only off by 1C. (From now on I'll also note manually the temperature every minute")
[image]

## 500 gr weight

Here I kept the offset for transparency, but we see that the measured temperature with the thermal camera reaches very similar values to the above ones.

[image]

[image]

pattacini, 2023-04-27:
What are the movements of the wrist in both cases, i.e., with and without load?
The patterns seem different.

mfussi66, 2023-04-27:
> What are the movements of the wrist in both cases, i.e., with and without load?

The pattern is mostly the same, a repeated yaw motion, +-40 deg and then rest at 0.

Continuing with the tests to make sure that the temperature significantly increases with the current, I applied 1Kg payload to the wrist. In the first plot, I repeated an alternate yaw motion, while in the second plot I kept the wrist still.
The results still seem to be poor,  but we can notice that the temperature estimation increases constantly due to the constant current (220mA), while the measured one settles at 41.5C
[image]

[image]

The thermal model flashed in the firmware, in its current form behaves just like the continuous one. Of course we can fit a model with the data, but maybe it's just a matter of reworking the transfer function.

mfussi66, 2023-04-27:
Reflecting together with @pattacini and @maggia80 we decided to:
-  Read the temperature with a thermocouple, to avoid any doubts regarding the thermal camera setting:
- Consider only the one of the two thermal resistances in the model

Using only the 7 K/W resistance at the moment seems to yield a small temperature difference, 1.7C after 15 minutes, compared to the 6.3 measured with the thermocouple

mfussi66, 2023-04-27:
A second test with 1Kg payload and 7 K/W resistance, gives the following results (in temperature delta) in steady state:
- Thermocouple 10.5 degC
- Thermal camera: 11.7 degC
- Thermal Model:  4 degC

There's a discrepancy also in this case, which suggest the necessity of estimating the thermal model.

mfussi66, 2023-04-27:
A side note: 

Since in the model $\Delta T_0 = 0$ and $T_{environment} = 25 C$ ( for example) we will have to be careful when power cycling the boards. We do not store the $\Delta T$ before turning off the boards, and therefore at the next boot the model will be reinitialized with the above values. However, the motor might still be warm enough to cause issues.

mfussi66, 2023-04-27:
> Reflecting together with @pattacini and @maggia80 we decided to:
> 
>     * Read the temperature with a thermocouple, to avoid any doubts regarding the thermal camera setting:
> 
>     * Consider only the one of the two thermal resistances in the model
> 
> 
> In the scenario of no payload applied and using only the 7 K/W resistance at the moment, seems to yield a small temperature difference, 1.7C after 15 minutes, compared to the 6.3 measured with the thermocouple.

Thinking about it, it would not make sense to use the 7K/W resistance, because if we used that we would have to take the temperature difference between windings  and housing, and we cannot measure the first. I think we should stick with the 49 K/W (or its fitted equivalent), because that's only where we are able to probe the temperature.

[image]

49K/W seems still too high given the measurements. As @maggia80 pointed out, the motor is not tested in the same conditions as ours, especially without gearbox. That could change the $R_{th2}$ estimate.

pattacini, 2023-04-28:
Let's proceed with identifying the parameters $R_t$ and $\tau$ robustly by covering a wide range of operating conditions with currents and temperatures (it'll certainly take a while to do so).

As usual, the validation will need to be performed on motors/setup/operating-conditions not seen during fitting.

closed 2023-04-28 by pattacini