robotology/icub-firmware-models #36: Fit the thermal model

pattacini, 2023-04-28:
Let's proceed with identifying the parameters $R_t$ and $\tau$ robustly by covering a wide range of operating conditions with currents and temperatures (it'll certainly take a while to do so).

As usual, the validation will need to be performed on motors/setup/operating-conditions not seen during fitting.

_Originally posted by @pattacini in https://github.com/robotology/icub-firmware-models/issues/34#issuecomment-1527158675_

pattacini, 2023-04-28:
We will be employing:
- the thermocouple, to measure the "absolute" starting and ending temperatures;
- the thermocamera, to measure the temporal evolution.

mfussi66, 2023-05-03:
Today I gathered some meaningful data after some trials, enough for a very first estimation of the model. Here is the data considering the computed dissipated power, and the measured temperature with the camera with the initial value removed.

[image]

I measured the temperature on the wires plastic connector, that would get significantly hotter than the metal cap. Moreover, the temperature oscillates significantly less.

I fitted the model with the training data, with the following result:

[image]

and validated on the validation data:

[image]

At the moment the model differs significantly from what we computed at first:

```
Process model with transfer function:
             Kp                      
  G(s) = ----------                  
          1+Tp1*s                    
                                     
        Kp = 11.981
       Tp1 = 402.55    
```

The results can definitely be improved by fitting with multiple experiments.

pattacini, 2023-05-04:
Some notes:
- Remember to ensure the model starts off from `0` in validation.
- I would have expected not a perfect match in validation, but a more minor discrepancy, honestly. This leads me to think that there might be aspects we're not catching with our simple thermal model.
- It's tedious to wait for such long test runs, but we also need to make sure that the exponential transition is complete and that we robustly reach for the steady state in our data.

mfussi66, 2023-05-04:
A second acquisition with the same conditions shows good repeatability and fit of the model, at least in the conditions of:
- 1Kg payload
- Yaw motion at 40deg/s
[image]

Currently I'm acquiring data with
- 1.5Kg
- No motion
- Longer settling time

so I can see the behaviour in a different scenario, and check if the previous model matches the following one.

mfussi66, 2023-05-04:
> Currently I'm acquiring data with
> 
> * 1.5Kg
> * No motion
> * Longer settling time (33min acquisition)
> 
> so I can see the behaviour in a different scenario, and check if the previous model matches the following one.

Two acquisitions mentioned above confirm the successful repeatability of the experiment in different conditions.

Here is the resulting plot, with RMSE of 0.6 degC.

[image]

However, we can notice even by the naked eye that there will be  differences in the two models estimated in two different conditions. Summarizing them here:

$$
\frac{\Delta T}{P} (s)= \frac{R_{th}}{\tau s + 1}
$$

||1Kg Yaw|  1.5Kg No yaw|
|---|---|---|
|tau|484.9| 402.6|
|Rth|15.63|10.46|

Maybe this discrepancy could be caused by not reaching exactly steady state conditions in the first case.

pattacini, 2023-05-04:
The step responses of the two systems above:

[image]

mfussi66, 2023-05-04:
Running the experiment with
- 1Kg
- no motion

Results in a model that gets closer to the one found with 1.5Kg:

||1Kg No Yaw|  1.5Kg No yaw|
|---|---|---|
|$\tau$|**438.25**| 402.6|
|Rth|**12.67**|10.46|

[image]

We might need a scaling factor in input to the model, which could coincide with the rms value of the input signal (can be computed easily in simulink).

mfussi66, 2023-05-04:
> Going a bit ahead, here're some nice results from using `Moving RMS`:
> 
> [image]
> 
> The model: [**`test_rms.zip`**](https://github.com/robotology/icub-firmware-models/files/11398888/test_rms.zip).

I therefore try to:
- Compute the Id and Iq RMS values of an experiment with varying currents
- Use the computed RMS to fit the model
- Validate the model with a dataset in which the current is kept constant

mfussi66, 2023-05-05:
After adding the moving RMS block, I tuned its exponential weighing factor to 0.995 empirically, accounting for the fact that decay during repeated motion at 40deg/s should not reach zero.
Then, I exposed directly the computed power via CAN, avoiding extra computations on the host.

Here is the training set taken by moving the yaw with 1Kg payload:

[image]

The fit on the training data is of course very good (90%):

[image]

The model in this acquisition has a significantly higher time constant, but comparable Rth to the previous experiments:

```
        Rth = 14.345                  
       tau = 586.78
```

I wanted to validate the model by getting data with a stopped rotor, but the thermal camera software kept crashing 😓. 

Nonetheless, I got the initial and final temperature (respectively 27.7degC and 53.7 degC), together with the computed power (which is constant in this case, around 2.16W). So we can infer from these parameters at least the Rth for the validation set:

(53.7 degC - 27.7 degC) / 2.16 W = 12.03

12.03 is 17% different from 14.35. 

We can surely acquire even more data to get more robust conclusions, but I think the takeaway is that the RMS block is needed to get more realistic values.

pattacini, 2023-05-05:
Superb! 🚀 

> I tuned its exponential weighing factor to 0.995 empirically, accounting for the fact that decay during repeated motion at 40deg/s should not reach zero.

Is this tuning general enough for running pretty much in all operational conditions?

Also, I think you can move on and propose the addition of the RMS block to the model via a PR.

mfussi66, 2023-05-05:
I think it depends on how the input signal varies. Given the application on the wrist at the moment, I think it's safe to keep the decay rate relatively quick to account for the types of sudden movements that the robot can make.
We can explore a range of waves (even synthetically) to verify my suppositions.

mfussi66, 2023-05-08:
> Also, I think you can move on and propose the addition of the RMS block to the model via a PR.

Landed here: https://github.com/robotology/icub-firmware-models/pull/37

mfussi66, 2023-05-08:
> We can surely acquire even more data to get more robust conclusions, but I think the takeaway is that the RMS block is needed to get more realistic values.

Further trials today showed that there is still a difference between model and validation set, when the latter is taken with a moving wrist. 

# Constant current

## Same input current
We have a good match when both training and validation set have a constant current, with two very similar amplitudes.

[image]

[image]

## Different input current

This good match occurs also when validating against current with a different amplitude (as much as the wrist allows):

[image]

[image]

In this case the model has: **Rt = 10.2 tau=384**

# Periodic current

Here the datasets were taken by moving the yaw repeatedly:
[image]
[image]

In this case the model has: **Rt = 17 tau=495**

# Training with constant current, validation with periodic current

In this scenario the difference is significant, as one can imagine, since the Rt differ a lot. 

[image]

A significant role in this difference is probably played by the 3 factors:
- downsampling of the current
- Rms value computed in a window of time
- Current waveform

pattacini, 2023-05-08:
We agreed to:
- Update the dictionary to rely on the params that came out from identifying against the constant current.
- Run the RMS within the FOC to gauge the effect of sampling.

pattacini, 2023-05-08:
>- Run the RMS within the FOC to gauge the effect of sampling.

Once in this condition, we may need to retune the Moving RMS params to get a more stable estimation of the power.

mfussi66, 2023-05-10:
I created a bespoke model to play around with different sampling times and weighing factor $\lambda$ of the moving RMS block.

The currents Iq and Id belong to the experiment with moving wrist, acquired at 10Hz and resampled (through interpolation) at 26.6KHz to emulate the FOC rate. Having them, I computed the dissipated power

[image]

I tried the following combinations:

|Ts| $\lambda$ |
|---|---|
| 26.6 KHz | 1 (true RMS at t->inf) | 
| 26.6 KHz | 0.995 | 
| 26.6 KHz | 0.99999 | 
| 100 Hz | 0.99 | 
| 100 Hz | 0.995 | 
|100 Hz | 0.999 | 
|10 Hz | 0.995 | 

The resulting plot at the moment is a bit chaotic:

[image]

So let's zoom on two relevant waveforms:

### Step signal
[image]

### Periodic signal
[image]

We can see that:
- At the FOC rate, $\lambda$ needs to be significantly increased compared to when running it at 1e-2
- Using the same $\lambda$ at 100Hz or 10 Hz changes significantly the settling time and the difference compared to the true rms

pattacini, 2023-05-10:
Nice statistics 👍🏻 
We can comment on them F2F.

Just one important note though.

>The currents Iq and Id belong to the experiment with moving wrist, acquired at 10Hz and resampled (through interpolation) at 26.6KHz to emulate the FOC rate.

I don't think this works ok. Upsampling a signal doesn't increase its real frequency content.

mfussi66, 2023-05-10:
> I don't think this works ok. Upsampling a signal doesn't increase its real frequency content. 

Of course, the resampling was just to establish a baseline and how the same lambda changes the results.

mfussi66, 2023-05-11:
Me and @pattacini checked F2F the plots in the simulation above and agreed to keep the computation of the RMS in the FOC with $\lambda = 1$ , relying on the fact that the time constant of the temperature is slow enough to allow for precise RMS evaluation.

mfussi66, 2023-05-11:
Here I ran the same motion with lambda=1 in the FOC, to qualitatively show how fast is the moving RMS response is. It takes 500 seconds to go from 0.3W to 0.1W, which is a time constant that is higher than the estimated thermal model, so it might affect the temperature estimation.

[image]

Nonetheless, I can try some acquisitions.

pattacini, 2023-05-11:
> a time constant that is higher than the estimated thermal model, so it might affect the temperature estimation.

I wouldn't be worried in this case as there's a temperature drop here rather than an increase.

pattacini, 2023-05-14:
Hi @mfussi66 

I've moved the RMS computation inside the FOC via 2b1c93cc3b739fb1b6945ff3a9a3fadbe4f02eb3.

mfussi66, 2023-05-16:
While getting more data to verify the fitting accuracy, I noticed that the evaluation of the RMS currents did not match my expectations in some conditions.
These conditions involve switching the current waveform, e.g. from:
1. variable (due to the moving wrist)
2.  to constant
3. and back to variable (due to the moving wrist)

Since $\lambda=1$, the RMS inside the code gives equal weight to all the samples, and simply accumulates them.  We expect RMS(point 1) == RMS (point 3), but since we switched to a constant waveform halfway, the accumulator is now polluted. 

I made a Simulink model to test this hypothesis, in which I periodically switch from constant to variable current (sine wave + noise), and it seems to confirm it:

[image]

With $\lambda=1$ the RMS value is never able to reach $1/\sqrt{2}$ as expected. Even worse, its derivative goes to zero and the value settles to a lower value than expected:

[image]

This is a consideration that [in this previous comment and test](https://github.com/robotology/icub-firmware-models/issues/36#issuecomment-1542445469) it was probably much harder to notice.

My suggestion is to keep $\lambda < 1$, such as 0.9995, taking into account the limitations of `single` datatype.

pattacini, 2023-05-16:
Somehow, I cannot click on images for inspection.

mfussi66, 2023-05-16:
Speaking with @pattacini , we agreed to keep the lambda=0.9995 and also acquire datasets at different speeds, to account for the fact that a constant current regime might be a limit condition.

mfussi66, 2023-05-16:
I got three datasets, trying to vary the speed as much as the wrist allows (but I can take more for better cross validation).
I fitted the model with one and validated with the other two, with very good results: 

[image]

In both cases the RMSE is very good < 0.5 degC.

Though the estimated parameters are significantly different compared to the constant current case:

Rt = 16
tau = 797.5

and they will overestimate the temperature (which could be good, since it would be a cautious measure). This can be seen in the lines in magenta, in which the estimated temperature is 4 degC higher than the measured one.

pattacini, 2023-05-17:
Nice result!

Let's put the case of constant current aside for a moment.
We should:
- Consider which **equivalent critical temperature** to account for to keep the windings safe (it'll be lower than that one in the datasheet). Probably, run a few tests for sanity check.
- Make a PR to update Rt, tau, lambda, Tcrit.

mfussi66, 2023-05-19:
> Instead, to complete this issue, we do just need to update the values of Rt, tau, lambda (and Tcrit?).

PR here: https://github.com/robotology/icub-firmware-models/pull/43

closed 2023-05-19 by pattacini