robotology/icub-firmware-models #40: AMC-BLDC: the supervisor needs to manage the motor-hal errors

valegagge, 2023-05-12:
Following the work done here https://github.com/robotology/icub-firmware/issues/360#issuecomment-1544022704, we want to update the supervisor in order to manage the motor-hal errors: `hardware-overcurrent` and hall-sequence-error.

**Dod**
1. The supervisor is updated
2. The code has been generated
3. The code has been tested.

Note: the test is not simple because this error are hw-related, so we need to find a way to simulate them.

pattacini, 2023-05-12:
There's also a connection to #15.

valegagge, 2023-05-12:
> Hi @valegagge
> 
> Just for clarification, in which manner the `hardware-overcurrent` differs from the overcurrent limitation we have implemented in the architectural model?
> 
> Is there a low-level check?

The driver calls this callback in case of faults and [here](https://github.com/ale-git/icub-firmware/blob/b05d579d29d359b9339e821cdb7f43ecb2951f7f/emBODY/eBcode/arch-arm/board/amcbldc/application/src/motorhal2/pwm.c#L424) sets the overcurrent flag. For this, I'm referring to an hardware-overcurrent, while in the supervisor, we check if the filtered current goes beyond the configured limit.

[image]

valegagge, 2023-09-19:
Checking the current code of the AMCBLDC board, the External fault is managed by polling instead of on ISR. So I think to implement the same solution for the hardware over current.

See [here ](https://github.com/robotology/icub-firmware/blob/bd6fdf188c1e47700a679ba3fa05858999532836/emBODY/eBcode/arch-arm/board/amcbldc/application/src/app-board-amcbldc/embot_app_board_amcbldc_theMBD.cpp#L57)and [here](https://github.com/robotology/icub-firmware/blob/bd6fdf188c1e47700a679ba3fa05858999532836/emBODY/eBcode/arch-arm/board/amcbldc/application/src/app-board-amcbldc/embot_app_board_amcbldc_theMBD.cpp#L412C4-L412C25).

Moreover, I verified that in the 2FOC fw, the hardware over current and software over current are notified to the EMS using the same bit in the status message. Therefore I'll adopt the same solution.

In the future, if we want to distinguish the two faults we need to change the can protocol.

I suppose that the callback `pwmMotorFault_cb` registered [here ](https://github.com/robotology/icub-firmware/blame/bd6fdf188c1e47700a679ba3fa05858999532836/emBODY/eBcode/arch-arm/board/amcbldc/application/src/motorhal2/pwm.c#L672) works properly.

pattacini, 2025-07-17:
Fine with me 👍🏻 
Leaving the last word to @mfussi66.

closed 2025-07-18 by valegagge