robotology/icub-firmware-models #42: Refine thermal model with inductive load

pattacini, 2023-05-19:
A F2F chat with @pattacini and @maggia80 highlighted that it is important to consider how the PWM signal is generated (whether it is in Left-aligned mode or Center-aligned), and how the current is measured.
Since we are dealing with an inductive load, we need to consider exponential transitory effect, during which the PWM signal of a phase is positive.
The current is measured in the middle of the positive PWM, during which the transitory might not be exhausted. This depends on the time constant (Phase resistance and inductance), and on the duty cycle. 

Therefore accounting for the electrical transitory period might be the key to improve the model and reduce the performance discrepancy in continuous current (aka Iq in the case above).

Support material:
 - [UNDERSTANDING THE EFFECT OF PWM WHEN CONTROLLING A BRUSHLESS DC MOTOR](https://www.portescap.com/-/media/project/automation-specialty/portescap/portescap/pdf/whitepapers/wp_understanding_the_effect_of_PWM.pdf)

_Originally posted by @mfussi66 in https://github.com/robotology/icub-firmware-models/issues/36#issuecomment-1553154677_

pattacini, 2023-05-19:
Hi @fgarini,

We would need your help to get to know how the PWM is actually generated in our AMCLBDC board to better organize in view of the [OP](https://github.com/robotology/icub-firmware-models/issues/42#issue-1716859081).

cc @andreaderito @maggia80 @Nicogene @mfussi66

fgarini, 2023-05-19:
## [WIP COMMENT]

<details><summary>Details</summary>
<p>

Ciao, in the amc_bldc the pwm generation is center-aligned. 
[image]
[image]

The ADC in use is the 2nd and the conversion is triggered by tim1 6th channel, oc2ref.
(tim1 configuration)
[image]
[image]
(adc2 configuration)
[image]

The ADC  has a silicon bug that requires to execute two consecutive conversions and discard the second one, for each phase.

[image]

I remain at your disposal.
Other useful information is written by Giorgio in the user manual contained in the repo `amc_bldc\docs`

</p>
</details>

fgarini, 2023-05-19:
I  completed the comment above. Let me add a few considerations: the motor IIT code `15562` has a quite high L/R constant with respectively `3,15mH` and `25.9Ohm` = `122uS`
the PWM period is (1/82kHz) = `12uS`.

Even in short circuit with a full-period PWM applied, the actual delta in the current current would be:

```math

I=I_{0}(1−e^{−t./τ}) \\

```

With 42V and 25.9Ohm, `I0` is `1.62A`
The `dI` is: [`1.62A` * (1-e^{-12uS/122uS)] = [1.62A * 0,094] = `0.15A`.
I did very conservative the calculations, since i put the condition of short circuit simultaneously applying the maximum charge voltage for the whole sample period.

In addition to that I don't find critical the settling time/step response of the current sensor ACS70331, that is nominally less than 550ns 
[image]