robotology/icub-firmware-models #45: Update Wrist-Mk2 model – Physical components

pattacini, 2023-05-19:
We're required to update the Simscape model of wrist-mk2 by exporting the latest mechanics from CREO.

cc @ale-git @salvi-mattia @valegagge @fiorisi @mfussi66 @Nicogene

ale-git, 2023-06-19:
I've imported the wrist model in Simmechanics, and I see that it is different from what we said here
https://github.com/icub-tech-iit/ergocub-design-lowerarm/issues/125#issuecomment-1431357633
being the angle between arm and platform joints is 60 deg and not 90 deg, thus its kinematics will be different.
[image]

@fiorisi @lawproto

pattacini, 2023-06-19:
cc @mfussi66

pattacini, 2023-06-19:
Hi @ale-git @lawproto

Remember that this is a public repo and links to private resources should be avoided.
It could be better then to continue the discussion in the original repo, if you think that you need to recruit that info.

Lawproto, 2023-06-19:
@pattacini it is not a big deal for me, but if it is for you I respect you. I just felt prompted to reply where I was summoned.

pattacini, 2023-06-19:
> @pattacini it is not a big deal for me, but if it is for you I respect you. I just felt prompted to reply where I was summoned.

No problem @Lawproto.
It was just a reminder.

ale-git, 2023-06-19:
Ok, I think that there is no real need to link the private repo - a picture from there will do the job.  Since I'm documenting here the model updating task I've to put here a note about the RoM that hasn't been yet analyzed since the analysis was made for 90 deg angle, and it has to be taken into account in the control logic.

ale-git, 2023-06-19:
Remark: now the motor enumeration is as follows. Anyway, since the mechanism can be mounted in different ways, we must remember to check the real order.
[image]

pattacini, 2023-06-19:
Hi @ale-git 

Does this mean that the order is different from the current one identifying M0, M1, and M2?
(I don't seem to find the relative picture – maybe @mfussi66 can remember where it is)

Lawproto, 2023-06-20:
@pattacini I checked with the mechanical assembly. For Mk2.0 it was like this:

[image]

To, me it would seem natural to do the same for Mk2.1.
[image]

But it is just my two cents 🪙 🪙 .

pattacini, 2023-06-20:
Thanks @Lawproto 👍🏻

>To me, it would seem natural to do the same for Mk2.1.

Yep, right, unless @ale-git has some constraints to deal with.
Just let's bear in mind that even a small think like changing the name convention may have impacts in many places.

ale-git, 2023-06-20:
@pattacini don't worry :) I assumed this order from the rest positions 0-120-240, but we can set any order we want by adding an offset. This reminds me that, since we'll have different versions of the wrist, the arm offsets mus be made configurable.

ale-git, 2023-06-20:
Hi @Lawproto , the CAD exported the vertical axis of rotation of the three pulleys pointing down as in the figure, so that the positive rotation is clockwise when looking from above. I reverted them in the simscape model in order to have CCW positive rotation as in the real joint but, if the CAD allows it, it would be better to have the three axis pointing up in order to be aligned to the robot control logic. I don't know if it is possible to impose a direction btw, since I remember that when we export urdf models the direction is random.
[image]

pattacini, 2023-06-20:
>I don't know if it is possible to impose a direction btw, since I remember that when we export urdf models the direction is random.

cc @mfussi66 @Nicogene

mfussi66, 2023-06-20:
> I don't know if it is possible to impose a direction btw, since I remember that when we export urdf models the direction is random.

It is kinda random (Creo has its own internal logic), but for generating the URDF it can be reversed by setting the csys in the `reverseRotationAxis` tag. So it's just a matter of exporting the first time, and checking if it's the desired direction. Otherwise, reverse it and export again.

fiorisi, 2023-06-21:
The Creo CAD model used to generate the SimScape model is [ic_021_g_134_mech.asm](https://github.com/icub-tech-iit/ergocub-design-lowerarm/blob/mech/wrist_stuff/ic_021_g_134_mech.asm.1).

ale-git, 2023-06-21:
> It is kinda random (Creo has its own internal logic), but for generating the URDF it can be reversed by setting the csys in the `reverseRotationAxis` tag. So it's just a matter of exporting the [...]

Yes, there is a further step in generating URDFs from the SimScape model where we can configure the script to invert joint directions when required, but we stop at the previous step when importing the model in Simulink.

ale-git, 2023-07-26:
I've updated the model with the new wrist Mk 2.1, and made the `wrist_decoupler` configurable at runtime. Now all the geometric parameters are accepted as input by the 'wrist_decoupler' subsystem.
They are:
- arm angle offsets
- platform angle offsets
- arm bending
- right/left wrist

ale-git, 2023-07-26:
This is the schema of the Simulink controller that makes use of the `wrist_decoupler` in the Simscape digital twin of the wrist.

[image]

ale-git, 2023-07-26:
Here the PR
https://github.com/robotology/icub-firmware-models/pull/56

closed 2023-07-26 by ale-git

ale-git, 2023-07-26:
I've updated the motion control library in icub-firmware, then I realized that this task wasn't scheduled yet.

https://github.com/ale-git/icub-firmware/commit/0847b2ab36cd3a4d5cab6f634c1952bbac672ab7