robotology/icub-firmware-models #50: Explore ways to support code reusability

pattacini, 2023-06-08:
A few possibilities to experiment with:
- https://www.mathworks.com/help/ecoder/ug/generate-reusable-code-from-library-subsystems-shared-across-models.html
- https://www.mathworks.com/help/rtw/ug/reusing-functions-in-generated-code.html

pattacini, 2023-06-14:
Following up on method 2, I've started off at https://github.com/robotology/icub-firmware-models/tree/feat/reusable-fcns.

Here's the first test:

[image]

With some fiddling, I managed to generate two reusable instances of the FOC component:

```c
void test_2foc_step(RT_MODEL_test_2foc_T *const test_2foc_M, ExtU_test_2foc_T
                    *test_2foc_U, ExtY_test_2foc_T *test_2foc_Y)
{
  DW_test_2foc_T *test_2foc_DW = test_2foc_M->dwork;

  // ModelReference: '<Root>/Model'
  control_foc(&test_2foc_U->Sensors, &test_2foc_U->FOCSlowInputs_f,
              &test_2foc_Y->FOCOutputs1, &(test_2foc_DW->Model_InstanceData.rtb),
              &(test_2foc_DW->Model_InstanceData.rtdw),
              &(test_2foc_DW->Model_InstanceData.rtzce));

  // ModelReference: '<Root>/Model1'
  control_foc(&test_2foc_U->Sensors, &test_2foc_U->FOCSlowInputs_f,
              &test_2foc_Y->FOCOutputs2, &(test_2foc_DW->Model1_InstanceData.rtb),
              &(test_2foc_DW->Model1_InstanceData.rtdw),
              &(test_2foc_DW->Model1_InstanceData.rtzce));
}
```

Promising preliminary results so far.
Keep on investigating.

pattacini, 2023-06-23:
I've opened the following PR:
- #53 

It's in draft as it needs to be tested first.

pattacini, 2023-06-27:
PR merged ✅

closed 2023-06-27 by pattacini

mfussi66, 2023-06-27:
PR in icub-firmware here: https://github.com/robotology/icub-firmware/pull/391