robotology/icub-firmware-models #58: AMCBLDC: supervisorTX doesn't handle the overcurrent flag.

valegagge, 2023-09-22:
While working on issue #40, I noticed that the SupervisorTX supervisorTX doesn't handle the overcurrent flag, while the SupervisorRX performs the check.

| **SupervisorTX**  | **SupervisorRX**  |
|---|---|
|[image]|[image]|
|As you can see, in the dictionary, Flafs struct doesn't contain the overcurrentinfo|The Fault handler state machine checks the filtered current value|

I suppose this is a bug, isn't it? @pattacini @mfussi66 

I can fix it before addressing the #40.

pattacini, 2023-09-22:
Hi @valegagge 

There are many flags in the STATUS that still need to be handled. See #15.
This is one of them.

Once the SV_TX is updated to handle overcurrent, then also the following function requires being updated:

```matlab
function pl = format_status_pck(control_mode, pwm_fbk, flags, shift, CAN_VOLT_REF_SHIFT)

pl = uint8((zeros(8, 1)));

pl(1) = uint8(control_mode);

pl(2) = uint8(0);

tmp1 = bitshift(pwm_fbk, CAN_VOLT_REF_SHIFT - shift);
tmp2 = typecast(tmp1, 'uint8');
pl(3) = tmp2(1);
pl(4) = tmp2(2);

pl(5) = uint8(flags.ExternalFaultAsserted);
pl(6) = uint8(0);
pl(7) = uint8(0);
pl(8) = uint8(0);
```

This function is embedded in https://github.com/robotology/icub-firmware-models/blob/master/boards/amcbldc/messaging/format_can_id.slx.

valegagge, 2023-09-22:
Hi @pattacini,
I know there are still many status flags to be handled (see #15 ), but this seems an oversight because I remember that some time ago we implemented this functionality. I was suspicious that you had removed it for some reason that I don't know since I'm not on this project for a long time. :)

Anyway, I proceed to fix this issue (taking into account your advice) and then continue with #40. 💪 

thanks!

pattacini, 2023-09-22:
> I was suspicious that you had removed it for some reason that I don't know since I'm not on this project for a long time. :)

Nope.
Yep, please, go ahead 👍🏻

valegagge, 2023-09-26:
I annotated here how to address the bug:
1. add the filed `overCurrent` into the `Flag` struct of the dictionary
2. update that field in the `MSG_STATUS_HANDLER`
3. update the former of the status message as @pattacini suggested in the [above comment](https://github.com/robotology/icub-firmware-models/issues/58#issuecomment-1731281403)
4. Perform tests of the supervisor.

pattacini, 2024-02-15:
As agreed T2T, I'll take over @valegagge for this task.

cc @Nicogene @mfussi66

pattacini, 2024-03-06:
Tests done and PR successfully merged.

closed 2024-03-06 by pattacini