robotology/icub-firmware-models #59: Extend data dictionary to deal with multiple motors – Stint 2

pattacini, 2023-09-28:


mfussi66, 2023-10-04:
I found a possible way to deal with element vectorization: using the scripting functions to work on the dictionary.

Currently, the Common prj contains `newboard.sldd`, which has what I would like to be common interfaces, enums and so on (I promise I will change its name). 
Now, I created a board specific dictionary, `amcfoc.sldd`, which references the one above. Inside this dictionary, I created `Actuators`, a Simulink.Parameter that can have its dimension set: 

[image]

However, its not possible to encode the constant "numberOfActuators" in the Dimensions entries, only its value. 

Creating a simple script such as: 
```matlab
% Get reference to design data of dictionary
dict = Simulink.data.dictionary.open("dictionaries/amcfoc.sldd");
dd = dict.getSection("Design Data");

% find desired number of actuators
numOfActuators = dd.getEntry("numberOfActuators").getValue;

entries_to_modify = ["Actuators"];

for i=1:length(entries_to_modify)

    % get reference to entry to modify the dimension
    e = dd.getEntry(entries_to_modify(i));
    % modify dimension and write it to referenced parameter
    simpar = e.getValue;
    simpar.Dimensions = [1, numOfActuators];
    e.setValue(simpar);

end
dict.saveChanges;
```

Allows to programmatically change the dimension of whatever parameters we want vectorized.

pattacini, 2023-10-04:
Awesome!

>(I promise I will change its name).

You know me 😉

mfussi66, 2023-10-04:
Latest commit (https://github.com/robotology/icub-firmware-models/commit/351f6d7ff2fc5817fcc07931da3f7dae4b31ba65) creates the Initial Configuration parameters, and stores them in the data dictionary. This is done in a less elegant way compared to the amcbldc, because I struggled to enforce the `ActuatorConfiguration` data type. Nonetheless, that can be vectorized as well. This array of structures can also be passed around to test models.

mfussi66, 2023-10-06:
Since I want to know how the code generator handles the initialization of multiple motors, I decided to start tackling also the issue of code generation. 
However, I needed some preliminary steps: with the help of @pattacini it was possible to enable the iCubTech code replacement library in the Common project so that it can be used by AMCFOC (I wrote a simple README.md in case the process needs to be repeated. See the folder [in my fork](https://github.com/mfussi66/icub-firmware-models/tree/amcfoc/boards/common/system/code_replacement)).

Now the code can be generated, but there is still the nasty limitation of the function-calls in the Motor Supervisor that cannot transform the blocks into `Reusable Function`. 
[image] See for reference this issue: https://github.com/robotology/icub-firmware-models/issues/61#issuecomment-1742658393

Therefore, I would modify the supervisor to simplify the function calls and get around the problem. Note that the modification would be temporary and with no care of the internal logic of the block, since it's not the focus of the test. Additionally, I need to replicate some settings that are in the Model Configuration of the AMCBLDC, namely how the variables are initialized. I don't want the motors configuration to be zero, but correspond to the desired initial values.

mfussi66, 2023-10-06:
I succesfully modified the supervisor so that it would become reusable. 
In the previous step, the generated code was in C. Setting code generation in C++11 in the model settings, however, triggers the following error:

[image]

Removing the _sharedutils folder has no effect.

It seems that comparing differences between settings, the sharedutils source is generated with C instead of C++:

[image]

Something that does not happen for the AMCBLDC. Maybe there is an additional setting that I need to find.

mfussi66, 2023-10-06:
> Something that does not happen for the AMCBLDC. Maybe there is an additional setting that I need to find.

Fixed it, as it was an issue with the parent model configured for C99. After fixing it, now it's necessary to set up the settings for concurrency, since there are issue with rate transitions and the C++ support:

[image]

pattacini, 2023-10-06:
Don't switch to C++.
This would add a complexity that is not required now.
Let's stick to C.

mfussi66, 2023-10-13:
After some effort on the Supervisor part I was able to generate the code for a For Each subsystem which embeds the whole motion controller. As you can see on the right we cycle through the input data (supervisor events, sensor data) depending on their dimension:
[image]

Now it's just a matter of studying it.

mfussi66, 2023-10-20:
In order to improve the configurability and vectorization of the data, I created three helper scripts that can automate some manual work:
1. [set_actuators_number.m](https://github.com/mfussi66/icub-firmware-models/blob/amcfoc/boards/amcfoc/system/set_actuators_number.m) : Reads from the amcfoc dictionary the number of motors (as a constant value) and applies it to all the design data elements that need to be vectorized
2. [actuators_configuration.m](https://github.com/mfussi66/icub-firmware-models/blob/amcfoc/boards/amcfoc/system/actuators_configuration.m) : contains all the initial configuration parameters for all motors; in this case just 2, but they can be arbitrary (just needs to create more elements)
3. [update_actuators_config.m](https://github.com/mfussi66/icub-firmware-models/blob/amcfoc/boards/amcfoc/system/update_actuators_config.m): opens the dictionary, calls `actuator_configuration.m`, and writes the configuration in the design data (of the dictionary, so that the code generation has the desired initalization values

Here you can see a demo usage:
I start off with an amc configured for only one motor, then I change the number of actuators I want, launch the scripts, and rerun the codegen, demonstrating the that now I have two instances of the initial parameters and the ForEach that runs with both:

https://github.com/robotology/icub-firmware-models/assets/38140169/eb668923-c577-49f6-97ef-987349346032

pattacini, 2023-10-20:
Awesome!

mfussi66, 2023-10-25:
The vectorization of input and output signals can be done somewhat seamlessly at the moment, one just has to set the number of elements in, for example, `SensorData` and `FOCOutputs` ports. What actually needs manual intervention at the dictionary level, i think, is inizialization parameters and stored parameters. The latter do not exist explicitly in our Simulink models, we keep memory through stateflow. 
Therefore, at the current control level I think we only need `Initconfparams` to be explicitly vectorized, while the rest is automatically handled. 
When we will deal with control at the joint level we might need more processing, but I think now this can be handled easily through the scripts.

mfussi66, 2023-10-26:
[image]

I was able to use the task scheduler to set up the rate of each task, like in the picture above. 

If the task scheduler is not set up, the code is called all in one giant looping function that makes it impossible to separate like in the amcbldc. 

There is that task at 1e-6sec that does nothing in its function body, and at the moment I do not know how to remove it. 

The interesting thing is that there is a ForEach iterator inside each task function, which is I think is a good first step to create our modular code.

pattacini, 2023-10-26:
>I was able to use the task scheduler to set up the rate of each task, like in the picture above.

Uhm, if I remember correctly, we don't rely on the task scheduler in the AMCBLDC, where instead the rates are assigned to the solver and the functions are still distinct and can be called properly from different places. The `ert_main`, of course, calls them from within one unique place.

>There is that task at 1e-6sec that does nothing in its function body, and at the moment I do not know how to remove it.

We have something similar in the AMCBLDC, so no problem 👍🏻

mfussi66, 2023-10-26:
> Uhm, if I remember correctly, we don't rely on the task scheduler in the AMCBLDC, where instead the rates are assigned to the solver and the functions are still distinct and can be called properly from different places.

Yes you are most likely right, it is only necessary to tick the option:
[image]
without explicitly configure the tasks.

pattacini, 2023-10-26:
Yes, indeed. The correct configuration of ticks and params needs to be carefully inspected.
My recommendation is that you open up a model of AMCBLDC to copy out all those settings, meaning all the configuration settings, for all options.

mfussi66, 2023-10-26:
In a F2F chat with @pattacini we agreed to proceed in a new issue, to investigate how to deal with the fact that the `FOCInnerLoop()` must be called twice, but by the interrupt, and not periodically. 
This one could be closed, since we gained precious insight on how to run the vectorization of signals, and how the codegen interacts with them.

pattacini, 2023-10-26:
Closing in favor of #66.

closed 2023-10-26 by pattacini