robotology/icub-firmware-models #60: AMCFOC – Refactor the Supervisor motor-independent part – Stint 1

pattacini, 2023-09-28:
Following up on the excellent work done on the dictionary (see #54), we need to deal with the motor-independent part of the Supervisor.

> [!important]
It's true that we aim to address the AMCFOC, but it's essential to keep in mind that at a certain time we'll have also to integrate the AMCBLDC CAN handler and 4 motors (AMC4DC) instead of 2.

mfussi66, 2023-11-30:
At the moment, the motor-independent part of the architectural model needs to send out the following data:

- ReceivedEvent
    -  EventType
    - Limits struct
    - MotorConfig struct
    - Targets struct
    - PID struct

So all the possible data is passed at once and depending on the event type the related struct is read.
A motor-independent block could implement different dispatching functionalities depending on the input (which is the output of the CAN Decoder in the AMCBLDC).

> Let's also remember that we need to check with a Sequence Diagram the timing of the interactions within the Supervisor.

Will do!

mfussi66, 2023-12-18:
> A motor-independent block could implement different dispatching functionalities depending on the input

Something along these lines:

[image]

The decoding part can be an interface to the inter-core communication, since I am not sure if it's convenient to send directly via ICC a N number of `ReceivedEvent` structures. 
It could be necessary to add a specialized decoding logic in that "Decoding" block above, similar to the CAN one.
 If not needed, we could most likely use the entry points of this new architectural model to attach to the communication logic and the rest of the firmware.

Maybe we could align in a meeting @pattacini @marcoaccame

pattacini, 2023-12-18:
My hunch is that we can get away without the `Decoding` block in the new architecture, meaning that one input to the main model will be represented by `Events` itself.

Therefore, we should make sure that this approach is fine for the AMCFOC and the AMC4DC.

For the AMCBLDC, either we will drop the CAN codec or we refactor it to make it compatible with the `Events` signal. Probably, the latter still has value since for a CAN board it could be useful to keep communication and control altogether in the mode.

mfussi66, 2023-12-18:
> My hunch is that we can get away without the Decoding block in the new architecture, meaning that one input to the main model will be represented by the signal Events itself.

I guess the only limit is the size of the ICC buffer, I can check the size of the `ReceivedEvents` struct at the moment to have a first approximation of the needed quantity.

pattacini, 2023-12-18:
> I guess the only limit is the size of the ICC buffer, I can check the size of the `ReceivedEvents` struct at the moment to have a first approximation of the needed quantity.

In my mind, between the data traveling over ICC and `Events`, there can be a further intermediate stage, which is fully FW-based, i.e., no model. Thus, compression can be still there.

pattacini, 2023-12-18:
The number of bytes seems quite high. However, optimizing now could be counterproductive. We will get back to this later.

Does this structure deal with a single motor or two motors?

pattacini, 2023-12-18:
>Sending the motor configuration every step and reading it only when needed might not be efficient, but we can change the behavior in the future if desired/needed.

Just for clarification, we don't send motor configuration every step. Nonetheless, memory allocation is persistent.

mfussi66, 2023-12-18:
If that's the case, we can then proceed to the other parts of the motor-independent supervision for the AMCFOC. I do not think much has changed since the original sketch of the architecture:

[image]

The next important component to tackle will then be the Kinematic Controller, tasked with implementing Position/Velocity control at the joint level.

Ideally, it would not be too dissimilar from the `Control_Outer` which is already implemented. The bare minimum changes would, I think, involve the reading of the joint encoder instead of the motor, and some state-machine to handle the joint control mode.

pattacini, 2023-12-18:
We agreed that the control of the joint will be inherited as done now in EMS/AMC, that is hand-coded.
Later, we will evaluate the possibility of extending MBD to cover this part as well.

As of now, we will be dealing just with the control of the motor _per se_, as done in the AMCBLDC>

mfussi66, 2023-12-18:
Ok 👍 This is a huge simplification, then!

So to recap, the supervisor-rx had these two components that would be considered motor independent:

- State handler
- Can message handler

[image]

I implicitly ignored them in this issue, because I think the new version of the motor-specific supervisor already has those blocks' features included, in some form.

## State handler

The state handler is made by:
- board state handler
- fault handler
- can messsages for loop

### board state handler

[image]

This FSM is already embedded in the Control Mode handler, and contains placeholders.

### fault handler

The fault handler was part of the motor-specific supervisor and was moved inside it.

### can messages for loop

This FSM is used for iterating over the 4 elements of the decoded can messages. This function is already implemented by the ForEach subsystem

## Can message handler

[image]

We are not really talking about can messaging at this stage, but the array of decoded messages, with the contents put in the proper structs. The FSM takes the array of structs and depending on the **can messages for loop** sends the data to the rest of the blocks.

This is handled in the new supervisor by this:

[image]

Each subchart basically sets the outputs of the supervisor and that's it, the rest of the motion controller reads them.

--------------

So in the end, I don't think we are missing important parts of the supervisor behaviour, what is needed is:
- restore the custom code calls
- create proper test to verify if all conditions turn out as expected

pattacini, 2023-12-19:
Closing in favor of #72.

closed 2023-12-19 by pattacini