robotology/icub-firmware-models #61: AMCFOC – Streamline block duplication

pattacini, 2023-09-28:
In https://github.com/robotology/icub-firmware-models/issues/54#issuecomment-1738923696, @mfussi66 found out that the For-Each approach is limited and cannot be applied straight to our case.

It'd be advantageous to poke around to see whether we can resort to methods better than our present solution based on block duplication.

mfussi66, 2023-10-02:
When I removed the calls to the custom C functions from the stateflow diagrams, I tried again to use the ForEach susbsys, but found the error: 
> 'Configuration Parameters > Model Referencing > Total number of instances allowed per top model' is set to 'Multiple' for model 'Supervisor'.  Subsystem block 'Supervisor/Setpoint handler' of this model is a function-call subsystem with a function-call port that has a width greater than 1. This port width is not allowed in a model configured to be referenced multiple times.

If I set the _Total number of instances allowed per top model_ = "One" I cannot use the ForEach block. 

So I made a toy model to make sure that it was the width of the callback signal causing problems, and indeed it is:

[image]

📦: [example.zip](https://github.com/robotology/icub-firmware-models/files/12781239/example.zip)

If doable, I would favour the refactor of the supervisor to eliminate multiple function-call subsystems such as the above.
In this way, we could better enforce the option of reusable functions and generate simpler code that is structured in the expected way.

pattacini, 2023-10-02:
>If doable, I would favour the refactor of the supervisor to eliminate multiple function-call subsystems such as the above.
In this way, we could better enforce the option of reusable functions and generate simpler code that is structured in the expected way.

Agree 👍🏻

mfussi66, 2023-10-26:
Updates to this issue were somewhat neglected, but similar work was done in [this issue](https://github.com/robotology/icub-firmware-models/issues/59#issuecomment-1781067191).
Indeed, there is the problem of the multiple function callbacks. If they are removed to keep just one of those, the code generation works for the ForEach subsystem. 
After discussing with @pattacini, we agreed of proceeding to refactor the motor-specific part of the supervisor, and close this issue since we identified the blockage of the streamlining.

closed 2023-10-26 by mfussi66