robotology/icub-firmware-models #66: Inspect and resolve the interaction between `For each subsystem` and FOCInnerLoop

mfussi66, 2023-10-26:
The FOCInnerLoop block is the one called by the ADC interrupt in the firmware. Since the for-each subsystem expect to call this function once for each motor every ~36usec, we need to check if there are any implementation mismatches between the codegen and the expected behaviour which is interrupt based and not periodic.

mfussi66, 2023-10-31:
The current codegen in C for the AMCFOC creates a series of functions for each referenced block. Each function has a for loop with the iterations given by the size of the input signals. 
The FOC function looks like this:

```C

/* Model step function for TID1 */
void mc_foc_tick(void)           /* Sample time: [3.65714285714286E-5s, 0.0s] */
{
  /* local scratch DWork variables */
  int32_T ForEach_itr;
  FOCOutputs rtb_ImpAsg_InsertedFor_FOCOutputs_at_inport_0;

  /* Outputs for Iterator SubSystem: '<Root>/Iterative Motion Controller' incorporates:
   *  ForEach: '<S1>/For Each'
   */
  for (ForEach_itr = 0; ForEach_itr < 1; ForEach_itr++) {
    /* ModelReference generated from: '<S1>/Model' incorporates:
     *  ForEachSliceSelector generated from: '<S1>/SensorData'
     *  Inport: '<Root>/SensorData'
     */
    motion_controllerTID1
      (&iterative_motion_controller_U.SensorData.jointpositions,
       &iterative_motion_controller_U.SensorData.motorsensors.Iabc[0],
       &iterative_motion_controller_U.SensorData.motorsensors.angle,
       &iterative_motion_controller_U.SensorData.motorsensors.temperature,
       &iterative_motion_controller_U.SensorData.motorsensors.voltage,
       &iterative_motion_controller_U.SensorData.motorsensors.current,
       &iterative_motion_controller_U.SensorData.motorsensors.hallABC,
       &iterative_motion_controller_B.CoreSubsys[ForEach_itr].Model_o1,
       &(iterative_motion_controller_DW.CoreSubsys[ForEach_itr].
         Model_InstanceData.rtb),
       &(iterative_motion_controller_DW.CoreSubsys[ForEach_itr].
         Model_InstanceData.rtdw));

    /* ForEachSliceAssignment generated from: '<S1>/FOCOutputs' */
    rtb_ImpAsg_InsertedFor_FOCOutputs_at_inport_0 =
      iterative_motion_controller_B.CoreSubsys[ForEach_itr].Model_o1;
  }

  /* End of Outputs for SubSystem: '<Root>/Iterative Motion Controller' */

  /* Outport: '<Root>/FOCOutputs' */
  iterative_motion_controller_Y.FOCOutputs_h =
    rtb_ImpAsg_InsertedFor_FOCOutputs_at_inport_0;
}

```

The function `motion_controllerTID1()` internally calls the `FOCInnerLoop` that we all know and love. So, we could call the former inside each interrupt. 

However, I think it's useful to first need to understand why SensorData is not explicitly a vector. It might be caused by the fact that in the model SensorData is just an input, and not explicitly an array of elements.

mfussi66, 2023-10-31:
UPDATE: I hid the previous message as the codegen did not seem to be triggered properly because I did not set a full rebuild. In fact the number of iterations was just 1.

The current codegen in C for the AMCFOC creates a series of functions for each referenced block. Each function has a for loop with the iterations given by the size of the input signals. 
The FOC function looks like this:

```C

/* Model step function for TID1 */
void mc_foc_tick(void)           /* Sample time: [3.65714285714286E-5s, 0.0s] */
{
  /* local scratch DWork variables */
  int32_T ForEach_itr;
  FOCOutputs rtb_ImpAsg_InsertedFor_FOCOutputs_at_inport_0[2];
  real32_T rtb_ImpSel_InsertedFor_SensorData_at_outport_0_motorsensors_[3];
  real32_T rtb_ImpSel_InsertedFor_SensorData_at_outport_0_jointposition;
  real32_T rtb_ImpSel_InsertedFor_SensorData_at_outport_0_motorsensor_0;
  real32_T rtb_ImpSel_InsertedFor_SensorData_at_outport_0_motorsensor_1;
  real32_T rtb_ImpSel_InsertedFor_SensorData_at_outport_0_motorsensor_2;
  real32_T rtb_ImpSel_InsertedFor_SensorData_at_outport_0_motorsensor_3;
  uint8_T rtb_ImpSel_InsertedFor_SensorData_at_outport_0_motorsensor_4;

  /* Outputs for Iterator SubSystem: '<Root>/Iterative Motion Controller' incorporates:
   *  ForEach: '<S1>/For Each'
   */
  for (ForEach_itr = 0; ForEach_itr < 2; ForEach_itr++) {
    /* ForEachSliceSelector generated from: '<S1>/SensorData' incorporates:
     *  Inport: '<Root>/SensorData'
     */
    rtb_ImpSel_InsertedFor_SensorData_at_outport_0_jointposition =
      iterative_motion_controller_U.SensorData[ForEach_itr].jointpositions;
    rtb_ImpSel_InsertedFor_SensorData_at_outport_0_motorsensors_[0] =
      iterative_motion_controller_U.SensorData[ForEach_itr].motorsensors.Iabc[0];
    rtb_ImpSel_InsertedFor_SensorData_at_outport_0_motorsensors_[1] =
      iterative_motion_controller_U.SensorData[ForEach_itr].motorsensors.Iabc[1];
    rtb_ImpSel_InsertedFor_SensorData_at_outport_0_motorsensors_[2] =
      iterative_motion_controller_U.SensorData[ForEach_itr].motorsensors.Iabc[2];
    rtb_ImpSel_InsertedFor_SensorData_at_outport_0_motorsensor_0 =
      iterative_motion_controller_U.SensorData[ForEach_itr].motorsensors.angle;
    rtb_ImpSel_InsertedFor_SensorData_at_outport_0_motorsensor_1 =
      iterative_motion_controller_U.SensorData[ForEach_itr].
      motorsensors.temperature;
    rtb_ImpSel_InsertedFor_SensorData_at_outport_0_motorsensor_2 =
      iterative_motion_controller_U.SensorData[ForEach_itr].motorsensors.voltage;
    rtb_ImpSel_InsertedFor_SensorData_at_outport_0_motorsensor_3 =
      iterative_motion_controller_U.SensorData[ForEach_itr].motorsensors.current;
    rtb_ImpSel_InsertedFor_SensorData_at_outport_0_motorsensor_4 =
      iterative_motion_controller_U.SensorData[ForEach_itr].motorsensors.hallABC;

    /* ModelReference generated from: '<S1>/Model' */
    motion_controllerTID1
      (&rtb_ImpSel_InsertedFor_SensorData_at_outport_0_jointposition,
       &rtb_ImpSel_InsertedFor_SensorData_at_outport_0_motorsensors_[0],
       &rtb_ImpSel_InsertedFor_SensorData_at_outport_0_motorsensor_0,
       &rtb_ImpSel_InsertedFor_SensorData_at_outport_0_motorsensor_1,
       &rtb_ImpSel_InsertedFor_SensorData_at_outport_0_motorsensor_2,
       &rtb_ImpSel_InsertedFor_SensorData_at_outport_0_motorsensor_3,
       &rtb_ImpSel_InsertedFor_SensorData_at_outport_0_motorsensor_4,
       &iterative_motion_controller_B.CoreSubsys[ForEach_itr].Model_o1,
       &(iterative_motion_controller_DW.CoreSubsys[ForEach_itr].
         Model_InstanceData.rtb),
       &(iterative_motion_controller_DW.CoreSubsys[ForEach_itr].
         Model_InstanceData.rtdw));

    /* ForEachSliceAssignment generated from: '<S1>/FOCOutputs' */
    rtb_ImpAsg_InsertedFor_FOCOutputs_at_inport_0[ForEach_itr] =
      iterative_motion_controller_B.CoreSubsys[ForEach_itr].Model_o1;
  }

  /* End of Outputs for SubSystem: '<Root>/Iterative Motion Controller' */

  /* Outport: '<Root>/FOCOutputs' */
  memcpy(&iterative_motion_controller_Y.FOCOutputs_h[0],
         &rtb_ImpAsg_InsertedFor_FOCOutputs_at_inport_0[0], sizeof(FOCOutputs) <<
         1U);
}
```

The function `motion_controllerTID1()` internally calls the `FOCInnerLoop` that we all know and love. So, we could call the former inside each interrupt. 

There might be unnecessary copies being performed, it's probably worth to investigate.

mfussi66, 2023-11-07:
> There might be unnecessary copies being performed, it's probably worth to investigate.

This is most likely caused by the necessity of the input function data to be passed around models with different rates. Not entirely sure that the copy can be solved in this case.

pattacini, 2023-11-08:
We could deem this inspection complete.
Well done @mfussi66 👍🏻

closed 2023-11-08 by pattacini