robotology/icub-firmware-models #72: AMCFOC – Test the architecture

pattacini, 2023-12-19:
Following up on https://github.com/robotology/icub-firmware-models/issues/60#issuecomment-1860980782, we're required to:
- restore the custom code calls
- create proper automated test to verify if all conditions turn out as expected

mfussi66, 2024-01-05:
The test was embedded in a Test Harness inside the supervisor, with sequence and assessment, as seen here:

[image]

Simply put, the **Test Sequence** sets the inputs to the supervisor in the usual sequence, and at the end it pushes the fault button. 
The step change occurs in the sequencer after 0.01secs, and the **Test Assessment** block checks if the condition for moving to the next step is satisfied.
For example, if at the `set_mode` step the previously set PID D value does not match (`check_pid_set`), a warning is thrown.

At the moment the checks are working as expected 👍  I can add more steps to verify if the transitions work as expected.

pattacini, 2024-01-05:
Nice!

> I can add more steps to verify if the transitions work as expected.

Ideally, we should check as many conditions and pathways as possible.

Another important point is to verify whether transitions occur in the expected amount of time without taking up longer lapses.

mfussi66, 2024-01-12:
I started adding a scenario to check for behaviour of hardware faults, so far there is only the overcurrent:

[image]

And it was useful because it made me realize that I forgot to reset the flag `isInFault`, which would allow to switch back to idle when the fault condition disappeared:

[image]

I'm not a fan of how I'm writing the test sequences, I'm curious to see if it can be avoided to repeat the steps when evaluating them in different conditions.

pattacini, 2024-01-12:
>I'm not a fan of how I'm writing the test sequences, I'm curious to see if it can be avoided to repeat the steps when evaluating them in different conditions.

Maybe barely related but still useful:
- https://www.mathworks.com/help/sltest/ug/using-an-external-function-in-test-sequence-block.html

mfussi66, 2024-01-15:
> > I'm not a fan of how I'm writing the test sequences, I'm curious to see if it can be avoided to repeat the steps when evaluating them in different conditions.
> 
> Maybe barely related but still useful:
> 
>     * [mathworks.com/help/sltest/ug/using-an-external-function-in-test-sequence-block.html](https://www.mathworks.com/help/sltest/ug/using-an-external-function-in-test-sequence-block.html)

I'll give it a look!

Today I implemented new checks, related to control mode switches and bumpless switch:

[image]

Also, I added a simple discrete linear model that simulates a RL circuit, so that it could be used as a very coarse emulation of a motor if more sophistication is needed

[image]

All in all the tests seem fine, but `check_mode_velocity` fails, because the output of the Supervisor occurs one sample after the input

[image]

mfussi66, 2024-01-19:
I found the culprit of the 1 sample lag above, a silly mistake:

In the input dispatcher, I was triggering the event of `SetControlMode` before storing the desired mode:

[image]

The event triggers the transition in the Control Mode handler who then checks for `requiredControlMode`. I thought the assignment was made before letting the Control Mode handler execute, who knew!

After swapping the two instructions, we have a perfect alignment of control mode changes:

[image]

pattacini, 2024-01-19:
Nicely debugged!

Tests are indeed useful!

mfussi66, 2024-01-22:
There was also a timing issue I found, that was caused by the `InputDispatcher`, and it was probably more critical. Basically, one FSM tick needed to occur before it could be able to process another event. 

To fix this, I replaced the subcharts in the dispatcher with graphical functions. See the picture bellow. In this way, the return occurs immediately and at the next tick the Supervisor is able to process another call.
[image]

mfussi66, 2024-01-25:
> Generating the static code metrics for ONE motion controller, we can see the following:

I noticed that I had a different compiler set up for the new supervisor, after setting the same one as the AMCBLDC, we have much better results:

|  | Old supervisor | New supervisor |
|--------|--------|--------|
| **Lines of code** | 1014 | 834 |
| **Global variables size (bytes)** | 664 | 0 | 
| **Accumulated stack size (bytes)** | 241 | 174 | 

I'm surprised by the zero on the global variables, maybe they are needed when many subsystems are interacting with each other.

mfussi66, 2024-01-25:
At the moment, I'm struggling to make the simulation work properly: it seems that even in open loop the simulated pmsm cannot spin:

[image]

I'll try to revert to a previous commit to see what changed.

pattacini, 2024-01-26:
> At the moment, I'm struggling to make the simulation work properly: it seems that even in open loop the simulated pmsm cannot spin:

Fixed via #77.

mfussi66, 2024-01-26:
Yep, and the new supervisor works as expected:

[image]

pattacini, 2024-01-26:
Nice!

pattacini, 2024-01-26:
Hi @mfussi66 

Just to make a recap of the task's objectives:
- <kbd>restore the custom code calls</kbd>
  Is this accomplished? 
- <kbd>create proper automated test to verify if all conditions turn out as expected</kbd>
  Lots of work done in this respect. Did you plan to extend the existing tests or can we consider this subtask complete?

mfussi66, 2024-01-26:
> Is this accomplished?

Done ✅ 
Both for the iCubTech code replacement library and the C functions called in stateflow

>  Lots of work done in this respect. Did you plan to extend the existing tests or can we consider this subtask complete?

I think It can be considered completed, also because the integration test in simulation was successful.

pattacini, 2024-01-26:
Fantastic job!
Closing the issue then.

closed 2024-01-26 by pattacini