robotology/icub-firmware-models #74: AMCFOC – Port latest dev from AMCBLDC project

pattacini, 2024-01-08:
This is a follow-up of #73 to address these points below:
- `Vcc` is now an input.
- CMSIS vs. STD variant (it's ok to keep the variant, eventually).
- Variant for the Thermal Model.

mfussi66, 2024-01-29:
Showing here the modifications I made to address this issue:

>     * `Vcc` is now an input.

Just added Vcc as a input as requested. Sensors is filled from outside the simulink model:

[image]

>     * CMSIS vs. STD variant (it's ok to keep the variant, eventually).

Done ✅

[image]

>     * Variant for the Thermal Model.

Similarly to the amcbldc, I created a variant configuration:

[image]

Which influences the RMS computation

[image]

And disables the temperature estimation:

[image]

The referenced model is inside the variant subsystem:

[image]

> 
> 
> cc @Nicogene

pattacini, 2024-01-29:
Nice!

mfussi66, 2024-01-29:
PR here! #79

pattacini, 2024-01-29:
Merged ✅

closed 2024-01-29 by pattacini