robotology/icub-firmware-models #76: AMCBLDC: Inspect suspect regression in simulation

mfussi66, 2024-01-26:
While working on [AMCFOC Testing](https://github.com/robotology/icub-firmware-models/issues/72) I tried to run a motor control simulation, using the latest `devel` branch. 

To my surprise, the simulated pmsm could not spin because there was no commutation between phases:

[image]

We can see that 3-phase currents and voltages remain somewhat constant, while they actually should have this kind of waveform:

Closed loop current control:
[image]

Open loop voltage control:
[image]

So I checked the AMCBLDC project, most specifically I used the testing model https://github.com/robotology/icub-firmware-models/blob/devel/boards/amcbldc/tests/control/control_pmsm.slx , and I noticed the same behaviour happening starting from the [PR regarding D-axis alignment](https://github.com/robotology/icub-firmware-models/pull/64), even though the physical motor spins with no problem.

I can make some tests to see which changes in this PR affect the simulation.

cc @Nicogene @pattacini @sgiraz

pattacini, 2024-01-26:
When I said that I used the PMSM model for a test lately, actually I only oversaw the current response, neglecting the speed. Therefore, this can explain why I didn't realize the problem[^1].

Changing the alignment types can be significant for the motor, indeed.

[^1]: Full tests are always helpful 😃

pattacini, 2024-01-26:
We need to change this, most likely:

[image]

mfussi66, 2024-01-26:
You might be right @pattacini , because if I switch back to Q-Axis alignment everything works.

pattacini, 2024-01-26:
I'm doing some tests for validation.

mfussi66, 2024-01-26:
> We need to change this, most likely:

Yep, that was it:

[image]

[image]

Thanks for the tip @pattacini !

pattacini, 2024-01-26:
Fixed via:
- #77

closed 2024-01-26 by pattacini

Nicogene, 2024-01-26:
It was fast 😆 
Great debugging indeed 🚀

PR #77 (2024-01-26): Fix axis alignment of the motor models
Fixed #76.