robotology/icub-firmware-models #80: Analyze ADC timing of AMCBLDC and AMC2C

Nicogene, 2024-02-02:
Let me try to recap what I got from a F2F alignment with @sgiraz.

1. There's pretty much NO difference between using STD or CMSIS ops on AMCBLDC and AMC2C.
2. On the AMCBLDC, the FOC is taking up $`8`$ $`\mu s`$ and NOT $`17`$ $`\mu s`$. The difference is represented by the ADC operations.
3. FOC takes up $`14`$ $`\mu s`$ on AMC2C, longer than on AMCBLDC.

- Point 1 completes this issue ✅ We go with STD. See #75.
- Follow-up question on Point 2: "Why does ADC last that long compared to FOC? Is there any margin for improvement?"
- Follow-up question on Point 3: "Why is AMC2C slower than AMCBLDC?" We know this is an ongoing study and we're probably at the right point to contact ST.

cc @Nicogene @sgiraz @marcoaccame @maggia80 @mfussi66

_Originally posted by @pattacini in https://github.com/robotology/icub-firmware-models/issues/67#issuecomment-1921377510_

Nicogene, 2024-02-02:
This task has to be done after harmonizing `Ts_FOC` between the 2 boards.

sgiraz, 2024-02-07:
In poi 2 I want to remark that the 8.3 us are taken to execute the `AMC_BLDC_step_FOC()` function when the control mode is `IDLE`.
When the control mode is set to `CURRENT` instead, it takes 10.7 us, see the scope plot below:

[image]

I Apologize for the mistake/confusion! 

By the way, the `EventViewer` shows that the whole DMA interrupt takes about 15 us:

[image]

Measuring with the scope:

.... ⚠️ in progress [image here] ⚠️

sgiraz, 2024-02-08:
### ➡️ AMC2C (66.6 KHz)

When the control mode is set to `CURRENT` instead, the `AMC_BLDC_step_FOC()` function takes $`20`$ $`\mu s`$, see the scope plot below:

[image]

The the whole `onCurrents_FOC_innerloop` function takes $`25.45`$ $`\mu s`$:

[image]

The whole `DMA` takes $`26.86`$ $`\mu s`$: 

[image]

cc @pattacini @marcoaccame @maggia80 @Nicogene

Nicogene, 2024-02-08:
To summarize we have

| | AMC2C (66.6 KHz) | AMCBLDC |
| :--- | :--- |  :--- | 
| `AMC_BLDC_step_FOC()` function | $`20`$ $`\mu s`$ | $`10.7`$ $`\mu s`$ |
| `onCurrents_FOC_innerloop` function | $`25.45`$ $`\mu s`$ | $`13.1`$ $`\mu s`$ |
| the whole `DMA` | $`26.86`$ $`\mu s`$ | $`14.34`$ $`\mu s`$  |

>  "Why is AMC2C slower than AMCBLDC?" We know this is an ongoing study and we're probably at the right point to contact ST.

A ~ 2x factor seems a lot, maybe it is the case of contacting ST

Nicogene, 2024-02-12:
The analysis of @sgiraz pointed out an important timing issue on `amc2c`, and this will be further investigate contacting ST, after that we hopefully know how to align the performance of the two boards.

Closing.

closed 2024-02-12 by Nicogene

pattacini, 2024-02-26:
Interesting timing for a comparison:

[image]

Source: https://www.mathworks.com/campaigns/offers/next/field-oriented-control-techniques-white-paper.html

> [!note]
We're not that bad, after all!

cc @marcoaccame @sgiraz @mfussi66 @Nicogene @maggia80