robotology/icub-firmware-models #85: Setting FOC Ts to 45us creates current oscillations

mfussi66, 2024-03-11:
...
    However, setting the FOC Ts to 45us makes the motor spin with more oscillations, is this expected?

[image]

_Originally posted by @mfussi66 in https://github.com/robotology/icub-firmware-models/issues/84#issuecomment-1983343711_

mfussi66, 2024-03-11:
This is a test with 45us . Note that with 65mA of target we get close to the no-load current.

[image]

Plots with 37.5us to follow asap.

mfussi66, 2024-03-11:
This is a test with 45us , and the AMC_BLDC architectural model. 

Note that with 65mA of target we get close to the no-load current.

[image]

Now with the 37.5us :  

[image]

At a first glance, the behaviour is very much comparable between the two. 

There's only an issue with -65mA, but it's probably due to a missing significant resistant load.

pattacini, 2024-03-11:
I think that we can reasonably consider the two outcomes very similar 👍🏻 
Thanks for the test, @mfussi66!

closed 2024-03-11 by pattacini