robotology/icub-firmware-models #98: Test the velocity control for the AMCFOC project

mfussi66, 2024-12-18:
The following is a diagram of the currently working modules on the AMC-FOC, tested successfully in simulation and on a HIL setup.
[image]

We are just missing a validation of the velocity control at the motor level (aka reading from quadrature encoder or hall sensors). Ideally, it should be validated first in simulation then on the HIL setup.

The aim of this issue is to track the tests and the tuning.

mfussi66, 2025-01-07:
cc @fiorisi @andreaderito

mfussi66, 2025-01-09:
I set the Velocity PI to 10, 10 and here is the first promising result. I sent via CAN a sine reference of 1000 deg/s of amplitude

Here is the Speedgoat side:

[image]

marcoaccame, 2025-01-09:
> 64 samples instead

at which delta time, 1 ms?

mfussi66, 2025-01-09:
> > 64 samples instead
> 
> at which delta time, 1 ms?

Exactly.

mfussi66, 2025-01-14:
Ah, the parameter `theshold.jntVelMax` was set to 1000, way too low. I increased it to 60000 deg/s which corresponds to 10000 rpm.

And now we have a motor that spins with no problem to 2500 rpm on the speedgoat, so the limit has effectively been lifted:

I can push it to a higher speed, but it depends on the maximum acceptable pwm in output.

Since the Vmax is 24V and Vcc is 48V, it is capped effectively at 50%. 

**Speedgoat:**

[image]

**AMC-FOC**

mfussi66, 2025-01-14:
With some tweaking on the supply voltage (to 96V) and definitely some compromises on the controller stability, we could even reach avout 45K deg/s, equivalent to 7500 rpm:

[image]

mfussi66, 2025-01-15:
Opened up a PR to merge the changes made during current control tests + velocity control tests in icub-firmware-models: https://github.com/robotology/icub-firmware-models/pull/99

closed 2025-01-17 by mfussi66