robotology/icub-firmware-models #108: MBD-MC-E - Study model for trajectory generator

marcoaccame, 2025-04-11:
## Task description 📑

The analysis in [here](https://github.com/icub-tech-iit/temp/issues/2) has shown that one the leaf objects in the `MCC` (Motion Controller in C) is the one represented by `Trajectory.c`.

This object is sued only by the `Joint` object and does not use external objects, so it seems a fair unit to model.

In this issue we want to explore how to model it in Simulink, how to test and how to generate code that can be added to the `MCC`.
In a later issue we shall deal with its integration.

### Useful links and documents 🔖 

See:
- https://github.com/robotology/icub-firmware/blob/devel/emBODY/eBcode/arch-arm/embobj/plus/mc/Trajectory.h
- https://github.com/robotology/icub-firmware/blob/devel/emBODY/eBcode/arch-arm/embobj/plus/mc/Trajectory.c

##  Definition of Done  🏁

To have a model, its test, its generated code.

pattacini, 2025-04-11:
Hi @RaffaelePumpo 

Have you considered this block?
- https://www.mathworks.com/help/uav/ref/minimumjerkpolynomialtrajectory.html

RaffaelePumpo, 2025-04-14:
Hi @pattacini ,

I've seen it and It might be a good idea to use it directly, it will follow in https://github.com/robotology/icub-firmware-models/issues/109. 
For now, I have updated like this : 

[image]

I've separated the process into different blocks: one for computing the coefficients and another for calculating the trajectory. This separation is done to maintain the structure of the existing functions in `Trajectory.c` as much as possible, since these functions are  used by several other modules. 

I'd like to gradually replace them with the ones implemented with the generated code, but without making drastic changes to the interfaces for now, but talking also with @marcoaccame we are going to analyze better the structure of the already existing c implementation, seeing if we can change the structure

pattacini, 2025-04-16:
This task can be probably closed as completed, @RaffaelePumpo @fgarini

closed 2025-04-17 by fgarini