robotology/icub-firmware-models #109: MBD-MC-E - define the deployment of the model for trajectory generator

fgarini, 2025-04-14:
## Task description 📑

in https://github.com/robotology/icub-firmware-models/issues/108 we studied the model for trajectory generator.
We want to deploy it in firmware in the proper way

##  Definition of Done  🏁

define how to deploy the model in firmware

RaffaelePumpo, 2025-04-15:
Following the analysis session yesterday with @marcoaccame , we reviewed all parts of the code where the functions defined in `Trajectory.c` are used. Based on this analysis, I implemented the models that replicate all the **core utility functions** of the `Trajectory` module.

[image]

The block calling te simulink functions are not essential, but I added them  to generate the code (otherwise, the corresponding C++ code wouldn't be created).

Once the model was built, I generated the C++ code and verified those.

Furthermore, I tested the functions `set_pos_end()` and `do_step()` to verify if the trajectory generation worked as expected:

[image]

[image]

---

### Issues to Resolve

1. In the generated code, the **`trajectoryhidParameters`** struct is **passed as a parameter** to every function.  
   We want these to become **class methods** instead, with the struct being the object itself.  

2. An additional **`trajectoryhidParameters`** struct is being passed as input-output in every function.  
   However, we expect the original `Trajectory` object to be **modified directly**, not by using a separate output parameter.
   Example of current code:

   ```cpp
      void Trajectory::Trajectory_velocity_stop(const trajectoryhidParameters *rtu_o,trajectoryhidParameters *rty_o1)
   ```

   Desired form:

   ```cpp
      void Trajectory::Trajectory_velocity_stop()
   ```
---

### Next Steps

- Explore **alternative trajectory generation methods** using existing **MATLAB blocks**.
- Investigate how these can be **integrated with current blocks** already in use.

mfussi66, 2025-04-15:
> 1. We want these to become **class methods** instead, with the struct being the object itself.

I think we would need to set up the data storage classes as in 

https://it.mathworks.com/help/releases/R2024b/ecoder/ug/config_code_defs_programmatically_example.html

mfussi66, 2025-04-17:
Awesome! We tended to rely on State machines to store state information and its initialization, this could be a promising pattern.

marcoaccame, 2025-04-18:
> Neat! We tended to rely on State machines to store state information and its initialization, this could be a promising pattern.

to find promising patterns is in our mission.

marcoaccame, 2025-04-18:
The way forward now is to model the API to what we need (hopefully minimal) that are not necessarily a mere copy of the current legacy `Trajectory` C object.

In #113 I have defined an `embot::app::mc::Trajectory` class that we can use all around the legacy C code. It has a simpler interface and is able to hide multiple implementations and to switch amongst them at compile time.

The first hidden implementation will be one containing the legacy `Trajectory` C object. 
The second will be the code you generate in here.

So, pls have a look at its API (with my help of course): 
- API: https://github.com/robotology/icub-firmware-models/issues/113#issuecomment-2812427905
- multiple PIMPL holder: https://github.com/robotology/icub-firmware-models/issues/113#issuecomment-2812702101

mfussi66, 2025-04-22:
@RaffaelePumpo is your work in a branch/fork ?

RaffaelePumpo, 2025-04-22:
Yeah it's here https://github.com/RaffaelePumpo/study-mbd-motioncontroller

mfussi66, 2025-04-22:
Thanks, is it possible to create a test harness in which the "class" functions are called so they can be tested? Or at least a second model that references Trajectory.slx . 

So that we can both generate code and run simulations with it.

RaffaelePumpo, 2025-04-22:
i’m going to try. For now i have created a model “test_trajectory.slx” with a stateflow that calls the functions.

RaffaelePumpo, 2025-04-22:
I was able to create the `test_harness` as requested @mfussi66, I have just done the push on Github . This test harness allows us to test the **Trajectory** functions simply by invoking the function directly from the **Stateflow** chart.

[image]

[image]

As a result, the previous model `test_trajectory.slx` is no longer needed and has been removed.
I have tested for trajectory generator giving a desired position = 2 and a velAv =2  getting : 

[image]
And also for trajectory generator giving a desired velocity = 1 and a accAvg = 4 getting:

[image]

### Trajectory Class 

All the necessary functions of **Trajectory** have been re-implemented to match those originally found in `Trajectory.c`. However, instead of procedural code, these are now encapsulated within a **class-based structure**, which groups all related functionality under clearly defined methods.
```cpp
namespace namespace_s1
{
  class Trajectory
  {
    // public data and function members
   public:
    // Block signals (default storage)
    struct B_Trajectory_T {
      float p;                         // '<S18>/Merge'
      float v;                         // '<S18>/Merge1'
      float a;                         // '<S18>/Merge2'
      int32_t return_value;            // '<S18>/Merge3'
    };

    // Block states (default storage) for system '<Root>'
    struct DW_Trajectory_T {
      trajectoryhidParameters Trajectory_h;// '<Root>/Data Store Memory'
    };

    // model step function
    void Trajectory_config_limits(const float rtu_pos_min, const float
      rtu_pos_max, const float rtu_vel_max, const float rtu_acc_max);

    // model step function
    void Trajectory_do_step(float *rty_p, float *rty_v, float *rty_a, int32_t
      *rty_value);

    // model step function
    void Trajectory_init(const float rtu_p0, const float rtu_v0, const float
                         rtu_a0);

    // model step function
    void Trajectory_set_pos_end(const float rtu_xStar, const float rtu_velAvg);
```

###  Next Steps

Following the analysis with @marcoaccame  last Friday, we agreed on the following next steps:

1. **Initial Validation**  
   We will test this first  implementation , verifying the generated code running on the **firmware**.

2. **Simplification & Abstraction**  
   After initial testing, we plan to **move away from the original C-based implementation**, simplifying the interface by reducing the number of exposed functions in **Trajectory**.  
   The goal is to leave only a few, user-friendly APIs, and replace custom trajectory calculations with equivalent **MathWorks-built blocks**, improving maintainability and readability.

mfussi66, 2025-04-24:
I suggest the next steps to be performed in a separate issue after the PR (including the test harness) has been done. I think, if the test harness is created, that you @RaffaelePumpo can create the PR.

marcoaccame, 2025-04-24:
> 1. **Initial Validation**
We will test this first implementation , verifying the generated code running on the firmware.

We have just run on the amcfoc board the generated code for the MBD trajectory. 
The code is in:
- https://github.com/marcoaccame/icub-firmware/tree/feat/mc-trajectory-00

And in here is a video:

pattacini, 2025-04-24:
>It is in https://github.com/RaffaelePumpo/study-mbd-motioncontroller

I suppose that the development is under review in #114.

`study-mbd-motioncontroller` no longer exists.

RaffaelePumpo, 2025-04-28:
Just to clarify, the version currently under review in #114 is the one that maintains the same interfaces as the original C code.
The one  I mentioned in https://github.com/RaffaelePumpo/study-mbd-motioncontroller is an improvement where I reduced the number of interfaces, but it has been only tested in Simulink for now, and we will consider if It is useful since the placeholder developed by @marcoaccame in https://github.com/robotology/icub-firmware-models/issues/113#issuecomment-2827748490 already does this "simplifications" .
Therefore, it is not part of the review currently ongoing.

fgarini, 2025-04-28:
Put the issue in review waiting for the approval of the pull request;
we will open new issue regarding implementation of the block and further minimum jerk polynomial and for firmware implementation.

marcoaccame, 2025-04-28:
Hi @RaffaelePumpo, just to say that this code is now in robotology's icub-firmware.

RaffaelePumpo, 2025-04-30:
PR approved in : https://github.com/robotology/icub-firmware-models/pull/114

closed 2025-04-30 by RaffaelePumpo