robotology/icub-firmware-models #113: MBD-MC-E - Prepare placeholder  inside the MC code for the Trajectory MBD code

marcoaccame, 2025-04-17:
## Task description 📑

Following the reordering of the code in #112  we can now prepare the placeholder in the MC code for the MBD generated code that will come from #109. 

In the past we added in the C code suitable macros that activate one piece of code or another all over the files. However, that makes difficult to understand and navigate the whole code.

The idea that we would like to follows is use a wrapper C++ class that incapsulates either the legacy or the new MBD implementation.
The class will have simplified API and use the PIMPL paradigm possibly with support of different implementations.

This phase for now is just for giving easy support and switchability to either the legacy or the MBD code for testing purpose. When everything works, eventually, we can use the MBD class directly. 

### Useful links and documents 🔖 

PIMPL
- https://en.cppreference.com/w/cpp/language/pimpl
- https://github.com/icub-training/material_modern-cpp/tree/master/exercises/day09/morepimpl

##  Definition of Done  🏁

We have PR merged in `icub-firmware.`

marcoaccame, 2025-04-17:
In here a tentative API of the class
```C++

/*
 * Copyright (C) 2025 iCub Tech - Istituto Italiano di Tecnologia
 * Author:  Marco Accame
 * email:   marco.accame@iit.it
*/

// - include guard ----------------------------------------------------------------------------------------------------

#ifndef __EMBOT_APP_MC_TRAJECTORY_H_
#define __EMBOT_APP_MC_TRAJECTORY_H_

#include "embot_core.h"

namespace embot::app::mc {
    
    class Trajectory
    {
    public:
        
        struct Point
        {
            float pos {0};
            float vel {0};
            float acc {0}; 
            
            constexpr Point() = default; 
            constexpr Point(float p, float v, float a) : pos(p), vel(v), acc(a) {}  
                
            void load(float p, float v, float a) { pos = p; vel = v; acc = a; }
            void clear() { load(0, 0, 0); }             
        };
        
        struct Setpoint
        {
            enum class Type { NONE, POS, VEL, POSraw, VELraw} ;
            
            Type type { Type::NONE };
            Point point {};
                        
            constexpr Setpoint() = default;
            constexpr Setpoint(Type t, float p, float v, float a) : type(t), point({p, v, a}) {}
            constexpr Setpoint(Type t, const Point &po) : type(t), point(po) {}
                
            void clear() { point.clear(); type = Type::NONE; }
                
            void loadPOS(float p, float withvel)    { type = Type::POS; point.load(p, withvel, 0); }
            void loadVEL(float v, float withacc)    { type = Type::VEL; point.load(0, v, withacc); }
            void loadPOSraw(float p)                { type = Type::POSraw; point.load(p, 0, 0); }
            void loadVELraw(float v)                { type = Type::VELraw; point.load(0, v, 0);  }
            bool isvalid() const                    { return Type::NONE != type; }
        };
        
        struct Config
        {
            float posmin {0};
            float posmax {0};
            float velmax {0};
            float accmax {0};            
            
            constexpr Config(float p0, float p1, float vm, float am) : posmin(p0), posmax(p1), velmax(vm), accmax(am) {}
            constexpr Config() = default;
            bool isvalid() const { return true; }
        };
        
                    
        Trajectory();
        ~Trajectory();
[... 84 lines cut ...]

// embot::app::mc::Trajectory::reset() as it is always used w/ 0 arguments
extern void Trajectory_init(Trajectory *o, int32_t p0, int32_t v0, int32_t a0);
// embot::app::mc::Trajectory::stop()
extern void Trajectory_stop(Trajectory *o, int32_t pos);
// embot::app::mc::Trajectory::stop()
extern void Trajectory_velocity_stop(Trajectory *o);

// marco.accame on 16apr2025: not used
// extern int32_t Trajectory_get_pos_ref(Trajectory *o);
// extern int32_t Trajectory_get_vel_ref(Trajectory *o);
// extern int32_t Trajectory_get_acc_ref(Trajectory *o);

// embot::app::mc::Trajectory::isdone()
extern BOOL Trajectory_is_done(Trajectory* o);

// embot::app::mc::Trajectory::get() for both of them
extern float Trajectory_get_target_position(Trajectory* o);
extern float Trajectory_get_target_velocity(Trajectory* o);

#endif  // include-guard

// - end-of-file (leave a blank line after)----------------------------------------------------------------------------

```

marcoaccame, 2025-04-17:
> At this point, the use of the legacy or the MBD implementation will be switched on / off just in one place inside file `embot_app_mc_Trajectory.cpp`.

... the choice of implementation at compile time on a PIMPL is not that trivial.

because the `struct Impl` is private to the class and cannot be easily assigned to be some other struct.

i have solved by defining possible multiple different implementation struct and then by deriving Impl from one of them, as in the following

```C++
// --------------------------------------------------------------------------------------------------------------------
// - choice of implemention
// --------------------------------------------------------------------------------------------------------------------

#if defined(useDUMMYimplementation)
struct embot::app::mc::Trajectory::Impl : public trajectory::dummy::Impl { };
#elif defined(useLEGACYimplementation)
struct embot::app::mc::Trajectory::Impl : public trajectory::legacy::Impl { };
#elif defined(useMBDv01implementation)
struct embot::app::mc::Trajectory::Impl : public trajectory::mbd01::Impl { };
#endif
```
`Code Listing`. The `struct embot::app::mc::Trajectory::Impl` is defined as the public derivation of another `Impl`.

In this way we can easily switch amongst a dummy, the legacy and the first version of the MBD generated implementation.

In here is the full `embot_app_mc_Trajectory.cpp` file.

```C++

/*
 * Copyright (C) 2025 iCub Tech - Istituto Italiano di Tecnologia
 * Author:  Marco Accame
 * email:   marco.accame@iit.it
*/

// --------------------------------------------------------------------------------------------------------------------
// - public interface
// --------------------------------------------------------------------------------------------------------------------

#include "embot_app_mc_Trajectory.h"

// --------------------------------------------------------------------------------------------------------------------
// - choice of implementation by assigning a macro
// --------------------------------------------------------------------------------------------------------------------

//#define useDUMMYimplementation
#define useLEGACYimplementation
//#define useMBDv01implementation

// --------------------------------------------------------------------------------------------------------------------
// - external dependencies
// --------------------------------------------------------------------------------------------------------------------

#if defined(useDUMMYimplementation)
// nothing is needed
#elif defined(useLEGACYimplementation)
#include "Trajectory.h"
#include "Trajectory_hid.h"
using LegacyTrajectory = Trajectory; // the one in Trajectory_hid.h
#elif defined(useMBDv01implementation)
// include the files from code generator
#endif

// --------------------------------------------------------------------------------------------------------------------
// - pimpl: private implementation (see scott meyers: item 22 of effective modern c++, item 31 of effective c++
// --------------------------------------------------------------------------------------------------------------------

// dummy implementation

#if defined(useDUMMYimplementation)
[... 176 lines cut ...]

bool embot::app::mc::Trajectory::tick()
{
    return pImpl->tick();
}

bool embot::app::mc::Trajectory::get(Point &target) const
{
    return pImpl->get(target);
}

bool embot::app::mc::Trajectory::isdone() const
{
    return pImpl->isdone();
}

bool embot::app::mc::Trajectory::stop(const float &v)
{
    return pImpl->stop(v);
}

// - end-of-file (leave a blank line after)----------------------------------------------------------------------------

```
`Code Listing`. The complete `embot_app_mc_Trajectory.cpp` file (still WIP).

marcoaccame, 2025-04-24:
In this fork:
- https://github.com/marcoaccame/icub-firmware/tree/feat/mc-trajectory-00

I have an `amcfoc.appl.yri` that:
- is able to use either the C legacy `Trajectory` or the new `embot::app::mc::Trajectory` upon exclusive definition of macros `MC_use_embot_app_mc_Trajectory` and `MC_use_Trajectory` inside file `EOemsControllerCfg.h`;  
- if `MC_use_Trajectory` is defined and `MC_use_embot_app_mc_Trajectory` is undefined, we must have the very same behavior as before because we exclude all the new code;
- else if `MC_use_Trajectory` is undefined and `MC_use_embot_app_mc_Trajectory` is defined we have activated a wrapped that can:
  - still call the C legacy `Trajectory` code
  - use the MBD generated trajectory using the model of  @RaffaelePumpo in its first version
  - add as many hidden implementations as we want so that we can test as many MBD generated trajectories as we want
  - add in the hidden implementation trace messages or time measurement code

marcoaccame, 2025-04-24:
The good news is that all the abive three modes work:
- C legacy `Trajectory` directly called,
- `embot::app::mc::Trajectory` w/ a PIMPL that calls the C legacy `Trajectory`,
-  `embot::app::mc::Trajectory` w/ a PIMPL that calls the MBD generated code.

A video is in here:
-

mfussi66, 2025-04-24:
Ideally, we should compare both by dumping yarp data and comparing reference trajectories with the angle feedback. We could do it in a separate issue.

marcoaccame, 2025-04-24:
> Ideally, we should compare both by dumping yarp data and comparing reference trajectories with the angle feedback. We could do it in a separate issue.

yes, i was discussing that w/ @RaffaelePumpo just a few minutes ago.

and also: compare the execution time of the main functions in both cases. for that i would use a dedicated issue because there are good practices to use. For instance w/ https://github.com/icub-tech-iit/knowledge-base/discussions/20

marcoaccame, 2025-04-24:
To reach the **DoD** and to close the issue I will merge this PR after I have tested that the basic behavior w/ these macros is the same as now. 

[image]

If so, it means that I did not do any cut and paste errors in selection of the code to use for trajectory and that the very same code as now runs.

later on we can do all the tests we need before we activate in robotology the first piece of the MBD-MC

marcoaccame, 2025-04-28:
The PR is ready:
- https://github.com/robotology/icub-firmware/pull/595