robotology/icub-firmware-models #115: MBD-MC-E - integrate Matlab 'trajectory plan' blocks

fgarini, 2025-04-28:
## Task description 📑
In https://github.com/robotology/icub-firmware-models/issues/109 the model of the trajectory generator has been deployed
We want to integrate the already existing Matlab blocks for computing the trajectory plan.

##  Definition of Done  🏁

- Explore alternative trajectory generation methods using existing MATLAB blocks.
- Investigate how these can be integrated with current blocks already in use.

pattacini, 2025-04-28:
Great for the ongoing work 👍🏻 

I've noticed that you've been using many MATLAB functions.
MATLAB functions need to be used only when strictly necessary and, judging from the names of those functions, I bet that at least some of them can have their proper Simulink implementation.

As a general comment, I can also see a kind of overcomplication of the approach (e.g., too many signals, too many blocks), when the functionality is pretty simple and already implemented in Simulink library. At any rate, this is an intermediate step and serves to bridge Simulink design with our handwritten codebase.

This is nice to get familiar with the methodology but can rapidly turn into a bottleneck as our main goal is to redesign the functionalities and the implementation from the ground up, no matter how the present codebase is working. Therefore, we shouldn't proceed with replacing each functionality one by one, copying what is already done.

This strategy surely applies to the core motion control functions, to sensors handling, calibration and so on. Protocols and the content of data packets travelling over comm networks can remain unchanged instead.  

I just wanted to make this point fully clear.
cc @mfussi66 @marcoaccame @fgarini @RaffaelePumpo

mfussi66, 2025-04-28:
> As a general comment, I can also see a kind of overcomplication of the approach (e.g., too many signals, too many blocks), when the functionality is pretty simple and already implemented in Simulink [...]

Totally agreed. Me and @RaffaelePumpo will work to streamline this small system , especially if we want to create a proper architectural model for the motion control of a joint. 

It won't be trivial to progressively propose replacements for components of the motion control, given how interconnected each module is. Intra (inter?) team strategizing will be crucial.

pattacini, 2025-04-29:
>Totally agreed. Me and @RaffaelePumpo will work to streamline this small system , especially if we want to create a proper architectural model for the motion control of a joint.

Great!

>It won't be trivial to progressively propose replacements for components of the motion control, given how interconnected each module is. Intra (inter?) team strategizing will be crucial to see if we can find a more holistic approach, but maintaining features (which are a lot).

Exactly, that was my point in the first place. Going all the way down through an iterative bottom-up approach would be quite painful, I'm afraid. We definitely need a substantial top-down vision where we rethink functionalities (and we would grab here the chance to make them more segregated, for example).

We can discuss this topic in more depth, given its importance.

mfussi66, 2025-05-02:
FYI @RaffaelePumpo you can use the tab "element assignment" in the data store read/write to target "class members: 

[image]

In this way a function like "set_pos_end" becomes more readable (especially the canvas gets smaller):

[image]

RaffaelePumpo, 2025-05-06:
> 2. **`coder_posix_time.h` missing**
>    Another header, `coder_posix_time.h`, is included in the generated code, but it is not present in any of the generated folders. After some investigation, it seems to be part of the MATLAB path, possibly related to internal use.
>    However, it appears that this include is **unnecessary** — none of the functions defined in that header are actually used.
>    ✅ _I resolved this by commenting out the include, and the code builds successfully without it._

I have found the file in `\MATLAB\R2024b\toolbox\eml\externalDependency\timefun`, it avoids future problems due to the previous comment out of the include of that file.
> 4. **Deployment issue on target hardware**
> I tried deploying the generated code to the board. Although the build process completes without any errors, once the binary is loaded, the board seems unable to switch to the application. It remains stuck in the eUpdater state, preventing me from actually testing the code. This could indicate a problem with the generated executable. 

Trying to debug the code we have noticed that the board is not able to exit from the eUpdater state since it goes to HW fault.
Specifically, after analyzing the generated source with @mfussi66, we noticed that the function implementing **minimum jerk block**, use `double` precision internally. These functions compute in `double`, and accept inputs as `double`, and then **cast results to `single`** for interfacing with the rest of the model — which operates predominantly in `single` precision (matching the hardware's expected format).
 This behavior could potentially lead to memory overconsumption. 
Here a function generated related to the block : 
```cpp
void Trajectory::MinimumJerkPlanner_mldivide_df(const double A[64], double B[8])
...

pattacini, 2025-05-06:
Nice investigation!
Point 3 and the fact that apparently there's no way to make it work with `single` would bring us back to the previous implementation.

RaffaelePumpo, 2025-05-12:
> Trying to debug the code we have noticed that the board is not able to exit from the eUpdater state 
   since it goes to HW fault.
Specifically, after analyzing the generated source with @mfussi66, we noticed that the function implementing minimum jerk block, use double precision internally. These functions compute in double, and accept inputs as double, and then cast results to single for interfacing with the rest of the model — which operates predominantly in single precision (matching the hardware's expected format).
This behavior could potentially lead to memory overconsumption.

Due to the deployment issues and potential resource/memory overhead introduced by the block's double-precision operations, we have decided not to integrate for now this block into the final firmware.

closed 2025-05-12 by RaffaelePumpo