robotology/icub-firmware-models #116: amcfoc - study rotor stator calibration in model

fgarini, 2025-05-12:
## Task description 📑

Study the modeling of rotor stator calibration in the AMC-FOC by the means of the qenc reading.

##  Definition of Done  🏁

Define the next steps to implement the rotor stator calibration

marcoaccame, 2025-05-13:
Hi all,
as promised in our today's meeting in here are some resources that are useful to understand what we mean with the term calibration of rotor-stator.

## Internal documentation

[iCub 3 Stator-Rotor auto-calibration - iCub Tech Docs](https://icub-tech-iit.github.io/documentation/icub_robot_calibration/icub3-stator-rotor-autocalibration/)

**comment**: we do only once the autocalibration procedure which is executed by the 2foc board which emits a CAN frame of type `ICUBCANPROTO_PER_MC_MSG__DEBUG` which is forwarded to YRI by the callback `eocanprotMCperiodic_parser_PER_MC_MSG__DEBUG(`) inside the par64 variable of a DEB_tag02 diagnostics message.

## internal posts

one in here where it is described how thisc alibration is done

[image]

[image]

- https://www.mathworks.com/help/mcb/gs/foc-pmsm-using-quadrature-encoder.html
- https://www.mathworks.com/help/mcb/gs/quadrature-encoder-offset-calibration-pmsm-motor.html
- https://www.mathworks.com/videos/field-weakening-control-for-pmsms-with-simulink-and-motor-control-blockset-part-1-field-oriented-control-calibrating-sensors-estimating-parameters-and-modeling-1618384749106.html
- https://it.mathworks.com/company/technical-articles/calibrating-optimal-pmsm-torque-control-with-field-weakening-using-model-based-calibration.html

## what it is according to copilot

[image]

[image]

so, basically: if we have the qenc we need to be sure that the measure we have is aligned to the magnetic field of the motor in order to do effective foc

marcoaccame, 2025-05-14:
Some more info from copilot:

> Q: explain to me please why in bldc motors with quadrature encoders is needed a calibration of the index position

In BLDC (Brushless DC) motors with quadrature encoders, calibration of the index position is essential for several reasons:

1. **Accurate Positioning**: The index position provides a reference point for the encoder, which is crucial for determining the absolute position of the motor shaft. Without this calibration, the motor controller would not know the exact position of the rotor, leading to inaccuracies in positioning and control.

2. **Synchronization**: Calibration ensures that the encoder's signals are synchronized with the motor's electrical commutation. This synchronization is necessary for the motor to operate efficiently and smoothly, as it allows the controller to correctly align the electrical phases with the rotor position.

3. **Error Correction**: Over time, mechanical wear and other factors can cause slight shifts in the encoder's position. Regular calibration helps correct these errors, ensuring that the motor continues to operate accurately and reliably.

4. **Initialization**: When the motor is powered on, the controller needs to know the initial position of the rotor to start the commutation process correctly. The index pulse provides this initial reference, allowing the controller to initialize the motor's position accurately.

5. **Improved Performance**: Proper calibration of the index position leads to better performance of the motor, including smoother operation, reduced noise, and increased efficiency. It ensures that the motor operates within its optimal parameters, providing better overall performance.

In summary, calibrating the index position in BLDC motors with quadrature encoders is crucial for accurate positioning, synchronization, error correction, initialization, and improved performance 

- [[1]](https://www.nxp.com/docs/en/application-note/AN1915.pdf)(https://www.nxp.com/docs/en/application-note/AN1915.pdf): [[3-Phase BLDC Motor Control with Quadrature Encoder using DSP56F80x]](https://www.nxp.com/docs/en/application-note/AN1915.pdf)(https://www.nxp.com/docs/en/application-note/AN1915.pdf)

- [[2]](https://electronics.stackexchange.com/questions/543613/encoder-vs-hall-effect-for-foc-control-of-bldc)(https://electronics.stackexchange.com/questions/543613/encoder-vs-hall-effect-for-foc-control-of-bldc): [[Encoder vs Hall Effect for FOC control of BLDC - Electrical](https://electronics.stackexchange.com/questions/543613/encoder-vs-hall-effect-for-foc-control-of-bldc)](https://electronics.stackexchange.com/questions/543613/encoder-vs-hall-effect-for-foc-control-of-bldc)
- [[3]](https://forum.digikey.com/t/guide-to-precision-encoders-from-quadrature-to-hall/45334)(https://forum.digikey.com/t/guide-to-precision-encoders-from-quadrature-to-hall/45334): [[Guide to Precision Encoders: From Quadrature to Hall - Motors](https://forum.digikey.com/t/guide-to-precision-encoders-from-quadrature-to-hall/45334)](https://forum.digikey.com/t/guide-to-precision-encoders-from-quadrature-to-hall/45334)

valegagge, 2025-05-16:
As discussee with @fgarini, I link here after the work carried on with @MSECode regarding the autocalibration. https://github.com/robotology/icub-firmware/issues/499

In few words, we experienced failures during the autocalibration on the new ergocub joints due to the different number of poles because the procedure has been validated for the current joints only.

mfussi66, 2025-05-16:
> * https://www.mathworks.com/help/mcb/gs/quadrature-encoder-offset-calibration-pmsm-motor.html

This looks like a pretty intuitive and simple way to calibrate the QENC: 

- Apply a Vd > 0 for a total of 10 seconds
- On the first 2sec, set the electrical angle to zero
- For the remaining 8sec, increase the electrical angle by generating it as a ramp in input to the inverse clarke-park, to "excite" the phases
- During this time i think it is assumed that the direct axes of rotor and stator will get aligned, therefore finding the offset.

RaffaelePumpo, 2025-05-19:
## Calibration in the 2FOC Board

### Initial Calibration (Performed at Every Power-On)

The initial calibration is performed every time the board is powered on and involves searching for the encoder index. This calibration consists of two main phases:

**Phase 1: Incrementing the Quadrature Current (Iq)**
- The quadrature current `Iq` is gradually increased until it reaches half of the nominal value (`Inom/2`).
- After reaching this value, a rotor alignment time is awaited.
- Subsequently, the "fake" encoder value (`encoder_fake`) is incremented.
- The process stops either if the **encoder index** is found or if `encoder_fake` exceeds 1.5 revolutions of the encoder (`ENCODER_1_5_REV`).
- If the index is not found within 1.5 revolutions, it proceeds to phase 2.

**Phase 2: Decrementing `encoder_fake`**
- In this phase, `encoder_fake` is decremented until the encoder index is found.
- If `encoder_fake` exceeds -1.5 revolutions of the encoder (`-ENCODER_1_5_REV`) without finding the index, an "index broken" error is reported.

### Full Calibration (To Find the Rotor-Stator Offset)

If a full calibration is desired to find the rotor-stator offset, additional phases are executed:

**Phase 3: Incrementing Iq to the Nominal Value**
- The quadrature current `Iq` is increased until it reaches the nominal value (`Inom`).
- The value of `encoder_fake` where the index was found is saved in a variable called `encoder_fake_0`.

**Phase 4: Measuring the Real Rotor Value (Delta A)**
- A delta (`deltaA` = 60 or -120 if found respectively in Phase 1 or 2 ) is added to `encoder_fake_0` value, and the real encoder value (`rotorA`) is measured.

**Phase 5: Measuring the Real Rotor Value (Delta B)**
- Another delta (`deltaB` = 60 or 0 if found respectively in Phase 1 or 2) is added to the `encoder_fake_0`, and the real encoder  value (`rotorB`) is measured.
- This results in two symmetric values around the index value, corresponding to two real values measured by the encoder.

**Calculating the Delta and Offset**
- The delta is calculated by taking the difference between `rotorA` (first rotor value) and `rotorB` (second rotor value).
- The delta is normalized to a range of -180 to 180 degrees.
- The offset is calculated as `offset = encoder_fake_0 + (deltaA + deltaB)/2 - (rotorA +delta/2) + 90`.
  - The 90-degree angle is added to obtain a more precise calibration offset, ensuring the field between the rotor and stator is maximized.

This full calibration process allows for a precise measurement of the rotor-stator offset, improving the system's accuracy.

RaffaelePumpo, 2025-05-21:
I have implemented a simple model to try the approach used in https://www.mathworks.com/help/mcb/gs/quadrature-encoder-offset-calibration-pmsm-motor.html explained before:
> This looks like a pretty intuitive and simple way to calibrate the QENC:
>
> * Apply a Vd > 0 for a total of 10 seconds
> * On the first 2sec, set the electrical angle to zero
> * For the remaining 8sec, [...]

It saves the value of the encoder after the exciting phase as offset, it's still not completely clear for me the reason,I think because it assumes that after that time the rotor has reached as specific value 0 (?) of the real angle of the motor , where it is fixed since the axes are aligned,  then it could explain why the offset is exactly the value of the encoder count.

## Doubts

1. I have tried also incrementing the value of the inertia and damping of the motor, It allows the rotor to stop in a position as follows:

[image]

Anyway changing the initial position of the rotor angle I thought that the angle of the motor where it assests  didn't change, but It changes.

2. Why it needs a ramp generator for the angle instead of a constant  angle?

RaffaelePumpo, 2025-05-23:
## Rotor-Stator Offset Calibration – Methods Overview

I’ve been reviewing different methods for calibrating the **rotor-stator electrical angle offset**.

Here’s a summary of three key approaches I’ve found:

### Method 1:

This method applies **constant voltage vectors `Vd`, `Vq`** with a known **constant electrical angle**, typically using 0° as a reference.

1. **Ramp phase**: A ramp is applied on the electrical angle (e.g., from `-π` to `π`) so the encoder catches the **index pulse**, ensuring it starts counting from the correct mechanical zero.
2. **Lock-in phase**: Then a **constant angle (e.g., `0°`)** is applied as electrical angle target, causing the rotor to settle to a known electrical zero.
3. At this point, the encoder reading directly gives the **rotor-stator offset**.

This method is used by (https://www.mathworks.com/help/mcb/gs/quadrature-encoder-offset-calibration-pmsm-motor.html)

However, in the implementation mentioned, **the ramp does not stop exactly when the index is detected**. Instead, it continues for a **random time**.
This works correctly **only if the motor has a single pole pair**, in which case there’s only **one electrical zero**.
But **for motors with multiple pole pairs**, there are **multiple electrical zero positions**, spaced across the full mechanical rotation. In this case, since the ramp continues past the index, the rotor may settle in **any of the sectors**, and the resulting offset will vary depending on where it stops. This means the calibration will **only point to the nearest electrical zero**, not necessarily a fixed one across all power-ups.

### Method 2

In this approach, a **ramp on the electrical angle** is applied, causing the rotor to rotate freely. By monitoring the **back-EMF**, and knowing that its amplitude is minimized or maximized at a specific electrical angle, the real electrical angle can be inferred.

Then, by comparing this inferred electrical angle with the **encoder’s mechanical reading**, the **offset** is computed.

This is described in (https://www.youtube.com/watch?v=M031gYuRoIg).

### Method 3

This method uses a sequence of **known voltage vectors `Vd`, `Vq`** with corresponding **expected currents `Id`, `Iq`**.

By observing the actual **phase currents** and transforming them to `Id`, `Iq` (e.g., via Clarke-Park), the difference between the expected and real current values is used to estimate the **angle offset**.

Reference: (https://ieeexplore.ieee.org/document/9376999)

### Conclusion

Out of the three, **Method 1** appears to be the **simplest and most robust** for our case, especially since it doesn’t depend on back-EMF or current sensing.

pattacini, 2025-05-23:
Well reported @RaffaelePumpo 👍🏻 

A few comments below.

>However, in the implementation mentioned, the ramp does not stop exactly when the index is detected. Instead, it continues for a random time.
This works correctly only if the motor has a single pole pair, in which case there’s only one electrical zero.
But for motors with multiple pole pairs, there are multiple electrical zero positions, spaced across the full mechanical rotation. In this case, since the ramp continues past the index, the rotor may settle in any of the sectors, and the resulting offset will vary depending on where it stops. This means the calibration will only point to the nearest electrical zero, not necessarily a fixed one across all power-ups.

I'm interested in checking the simulation and the results at the next silo meetup.

>Out of the three, Method 1 appears to be the simplest and most robust for our case, especially since it doesn’t depend on back-EMF or current sensing.

True that Method 1 seems the simplest but measuring currents is not problematic at all.
Method 2, instead, can be tough to realize as the motor should be driven externally. Also, measuring BEMF directly could be not that easy, probably.