robotology/icub-firmware-models #123: Allow switching between joint or motor encoder in position/velocity mode

fgarini, 2025-06-09:
## Task Description
TBD 

## Definition of Done

In progress

mfussi66, 2025-06-25:
I added an additional sensor input to the motion controller that can be set inside the 1msec tick, called `JointData`. The structure cointains only the joint position and it is passed around to the following components, as highlighted:

[image]

At the moment, I disabled the estimation of the joint velocity, and disabled the processing inside the supervisor to make smaller steps. The modifications in https://github.com/robotology/icub-firmware-models/issues/123#issuecomment-2990509297 can be easily added back. 
I will open a PR and run tests to check if behaviour was changed.

mfussi66, 2025-06-25:
The behaviour is unchanged, as we can see in the plot below the rotor speed is the same as the target and the testimated one.

**Legend:**
- current: target in A, output of velocity pid
- outputs.Iq_fbk: current feedback in A, output of clarke-park
- speed: actual motor speed in deg/s
- msgTx.foc.velocity: estimated speed in deg/s
- targets.velocity: speed target in deg/s
[image]

mfussi66, 2025-06-25:
PR here: https://github.com/robotology/icub-firmware-models/pull/132

closed 2025-06-25 by mfussi66