robotology/icub-firmware-models #135: Study the possibility of improving the AMCx architectural models to allow cleaner rotor-stator calibration implementation

mfussi66, 2025-07-09:
After successfully testing in simulation the qenc (rotor-stator) calibration, we noticed that we could improve the implementation by making improvements to the architectural model of the AMC( _/BLDC/FOC) boards. This step can be a first one towards the architecture proposed in https://github.com/robotology/icub-firmware-models/issues/134 . 

For example, we need to evaluate if it's beneficial to create a `Process Sensors` block outside of the motion controller, putting it on the toplevel architecture. Additionally, an important constraint, we would like to keep the inputs and outputs untouched.

RaffaelePumpo, 2025-07-11:
I propose inserting a new component called `Processing Sensors` between the `Motor Sensors` and the `Motion Controller`. This block would handle both the Event (required for the correct interpretation of sensor data, such as motor configuration) and the Raw Sensors:

[image]

Additionally, I suggest removing the offset interface from the `QuadratureEncoder` and instead including it in the `EstimatedData` structure. This would allow the offset to be passed directly to the estimation block, and from there to the supervisor. 
[image]

The supervisor could then:

1. Save the offset in the Actuator Configuration (in the case of a full calibration).

2. Output the offset outside the motion controller if needed.

Furthermore the inputs and ouputs of the motion controller remains untouched. 

As a result, the testing setup would look like this:

[image]

RaffaelePumpo, 2025-07-29:
Starting from the new structure of the Iterative motion controller done by @mfussi66  i have added the calibrator and process_sensors blocks as follows : 

[image]