robotology/icub-firmware-models #141: Add process sensors and calibration blocks

fgarini, 2025-08-05:
## Task description 📑
Add process sensors and calibration blocks
TBD @RaffaelePumpo 

### Useful links and documents 🔖 

##  Definition of Done  🏁

PR merged

RaffaelePumpo, 2025-08-07:
### AMCFOC 
I've integrated the sensor and calibrator blocks into the AMCFOC model it changes in this way:

[image]

### AMCBLDC
Here's the current state of the integration:

[image]

valegagge, 2025-08-11:
Hi guys,
I’d suggest that, also in the future, we consider implementing diagnostics for the encoder. We’ve seen that the QENC reading is not always perfect, and it can sometimes report a few counts more or less than expected. The Five team had previously built a [test bench](https://github.com/icub-tech-iit/study-encoders/issues/3) to validate the reading performed by the QENC firmware. Currently, both the CAN and Ethernet protocols already support transmitting information regarding the quality of the encoder reading, so we could leverage that for better monitoring and validation.

mfussi66, 2025-08-26:
Closing it since the DoD was satisfied.

closed 2025-08-26 by mfussi66